RavenSystem/esp-homekit-devices #1016: Special garage door command

tlagarto, 2020-06-15:
Could someone help see understand how do I get this special command to a garage door?
I want to instruct my sliding front gate as a "walk-in gate". I would be something like this:
Door is closed
1 ->start door (to open)
2->wait 9,5 seconds
3 ->start door (to stop)
4->wait 5 seconds
5 ->start door (to close)

This way I do not have to open the gate all the way and there's no need to press the button several times. It would be automatic.

This is the code that I'm trying to use but does not work as expected with my device (sonoff mini).
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[]},"1":{"r":[{"g":12,"v":1,"i":0.5},{"g":12,"v":0,"i":10},{"g":12,"v":1,"i":10.5},{"g":12,"v":0,"i":15.5},{"g":12,"v":1,"i":16}]},"2":{"r":[]},"3":{"r":[]},"t":40,"d":20,"f3":[{"g":4,"t":0}],"f4":[{"g":4}],"s":0,"b":[]}]}`

I'm sure I'm doing some wrong on the several instructions and wait periods, but I don't really understand what to do to fix this.
(I'm using the HAA Json Configurator)

Can anyone help?

csupi003, 2020-06-21:
My advice would be to check [Nold Open](https://www.nold.io/hu/compatibility/)  or the manufacturer’s instruction because your opener may support walk mode which means it has a dedicated connector for it. It can be set on the opener how far it goes. So it won’t go back automatically but only have to press it for open and close.

tlagarto, 2020-06-21:
Yes, I tried that option but, as you're saying it does not have a auto close timer. Hence my search for a Jason string that could mimic that...

tlagarto, 2020-06-25:
I think I got it!
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"r":[{"g":12,"v":1,"i":0.5}]},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"t":40,"d":18,"xa":0,"f3":[{"g":4,"t":0}],"f4":[{"g":4}],"s":0,"b":[]},{"1":{"s":[{"a":1}]},"i":1,"b":[{"g":0,"t":1}],"s":0},{"0":{"r":[{"g":12,"v":1,"i":0.5},{"g":12,"v":0,"i":10},{"g":12,"v":1,"i":10.5},{"g":12,"v":0,"i":15.5},{"g":12,"v":1,"i":16}]},"t":1,"b":[],"s":0,"i":0.5,"xa":0}]}`

3 buttons: garage door + walk-in + config in a sonoff mini (with dry contact adaptation)

closed 2020-07-25 by stale[bot]