RavenSystem/esp-homekit-devices #1019: Can not connect to the Home app.

lagger, 2020-06-18:
Hello. I've conifigured my Sonoff Basic R3 with this firmware. It connects to the WiFi but I can not connect it to my Home app. Have you experienced something like this? If yes then I beg you for the solution :D.

hyperexpert, 2020-06-18:
Have you added your JSON config? It won't connect to your Home app without a JSON config in place first.

lagger, 2020-06-18:
Yeah, I took one from examples and copy-pasted it. My JSON config looks like this:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

I just wanted a simple switch connected to Home app.

tlagarto, 2020-06-22:
I don't own a Sonoff Basic R3, can't tell if it works exactly the same way as the Sonoff Basic, but if it does, I found nothing wrong with your Json string.
I can add that with my Sonoff Basic every time I save a new Json, it it takes quite a few minutes (7 or 8) to show up on Home app. It might be due to updating FW, but I can't tell for sure.

If still does not show up, I'll config it again and wait. It eventually shows up. 
Once, I "gave up" and left it on. I came back 20min after and noticed it was working, don't know why or how, but it was.

Whiterabbit64, 2020-06-22:
Do you have one single SSID for both 2.4 and 5GHz devices ?
If yes, try to split into 2 SSIDS: one for 2.4GHz devices (on which you'll connect the Sonoff) and one for 5GHz devices.

closed 2020-07-22 by stale[bot]