RavenSystem/esp-homekit-devices #1022: Shelly RGBW with external Pusg

Hanimaniac, 2020-06-21:


CannyFoxx, 2020-06-23:
You can Home app for grouping your devices
[image]
If your device have free GRIO (usually it is GRIO0) - you can add physical button

Hanimaniac, 2020-06-24:
CannyFoxx,
thanks for your answer. 
If I Group with other Accessories the Hone App, I just have One Button to controll all of the grouped lights. 
In my second picture you can see i have 2 switches for two lights in one Accessoirie. 

in the following picture I found out that the button is. gpio13 but werent able to use it. 

[image]

greets
Hanimaniac

closed 2020-07-03 by Hanimaniac