RavenSystem/esp-homekit-devices #1025: Two issues in HAA last version.

WojtekWro, 2020-06-27:
Hi Jose,
I found two issues in the last version of your software.
First I faced in an accessory 45. After a lack of electricity accessory boots in status close instead last known position which was default in the previous version.
Second is connected with the itching time. I’ve set 10 seconds inching time in my switch. After this time relay goes to back to previous state (off) but in HomeKit it is still seen as “on”. 

Besides I love and appreciate your work. I’m really impressed. 
Thank you.

shanghaix, 2020-06-27:
I have the same issue with a shelly 25 not showing the last position after reboot, it shows close instead.

Kienz, 2020-06-30:
I have the same issue with a shelly 2.5 not showing the last position after reboot, it shows closed instead.

RavenSystem, 2020-07-05:
Window cover fixed with v2.5.0.

> Second is connected with the itching time. I’ve set 10 seconds inching time in my switch. After this time relay goes to back to previous state (off) but in HomeKit it is still seen as “on”.

You must declare inching at accessory root to apply to accessory status instead relay.

closed 2020-07-05 by RavenSystem