RavenSystem/esp-homekit-devices #1029: HTTP PUT request with custom headers

bvermes17, 2020-06-30:
Hello @RavenSystem ,

I am trying to create a heating&cooling thermostat device from an ESP-01S a temperature sensor and a relay. The relay controls the heater, this part is working great. My problem is that my cooler is controller with IR by another device which is added to homebridge but it is controllable with HTTP PUT requests. 

From a linux terminal, I can switch on my climate by issuing the following cURL command: 

curl -X PUT http://IP:PORT/characteristics --data '{"characteristics":[{"aid":8,"iid":10,"value":true}]} --header "Content-Type:Application/json" --header "authorization: AUTHKEY"'

In this command above of the data field two additional http headers are also passed which contain authorization key and content type. How could I create a config for this? This is the relevant part of my configuration but I cannot find where to pass the additional headers: 

      "4": {
        "h": [
          {
            "h": "IP",
            "p": "PORT",
            "m": 1,
            "u": "characteristics",
            "c": "{\"characteristics\":[{\"aid\":8,\"iid\":10,\"value\":true}]}"
          }
        ]
      },

I read through the documentations but could not find an answer. 

Thanks for the help in advance!

RavenSystem, 2020-07-05:
You must use TCP RAW request instead, and create complete payload by yourself.

bvermes17, 2020-07-06:
Hello @RavenSystem ,

Thanks for your reply. I tried creating the TCP request myself but without success. (captured the working packet with wireshark and replicated it but no success. )

So I went another way and used a simpler API for my IR blaster that can react to simple HTTP POST request. 

Command with cURL is working and turning on the climate: 

curl -X POST 'http://192.168.0.60:8000/blasters/name/mini3?target_name=ac&command_name=on'

Unfortunately no matter what I try I cannot make it work. Can you please have a look at my full config? Do you see the mistake why its not working? 

Thanks!

{
  "c": {
    "o": 0
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 0,
            "v": 1
          }
        ],
        "h": [
          {
            "h": "192.168.0.60",
            "p": "8000",
            "m": 2,
            "u": "blasters/name/mini3?target_name=ac&command_name=off"
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 0,
            "v": 1
          }
        ]
      },
      "2": {
        "h": [
          {
            "h": "192.168.0.60",
            "p": "8000",
            "m": 2,
            "u": "blasters/name/mini3?target_name=ac&command_name=off"
          }
        ]
      },
      "3": {
        "r": [
          {
            "g": 0,
            "v": 0
          }
        ]
      },
      "4": {
        "h": [
          {
            "h": "192.168.0.60",
            "p": "8000",
            "m": 2,
            "u": "blasters/name/mini3?target_name=ac&command_name=on"
          }
        ]
      },
      "t": 21,
      "b": [],
      "g": 2,
      "n": 3,
      "w": 3
    }
  ]
}

RavenSystem, 2020-07-06:
Try to use GET instead POST because you are not sending any content.

bvermes17, 2020-07-06:
@RavenSystem For a GET request its not responding. This URL has to be used with a POST HTTP. But just to check out, I added some dummy value as content in cURL it does not make a difference it turns on but with HAA even if I add content its not turned on. 

Do you have any other idea? 

This is my current config with the dummy content: 
`{"c":{"o":3},"a":[{"0":{"r":[{"g":0,"v":1}],"h":[{"h":"192.168.0.60","p":"8000","m":2,"u":"blasters/name/mini3?target_name=ac&command_name=off","c":"111"}]},"1":{"r":[{"g":0,"v":1}]},"2":{"h":[{"h":"192.168.0.60","p":"8000","m":2,"u":"blasters/name/mini3?target_name=ac&command_name=off","c":"111"}]},"3":{"r":[{"g":0,"v":0}]},"4":{"h":[{"h":"192.168.0.60","p":"8000","m":2,"u":"blasters/name/mini3?target_name=ac&command_name=on","c":"111"}]},"t":21,"b":[],"g":2,"n":3,"w":3}]}`

These are my logs from the device, when I turn the cooling on some HTTP request happens but the port seems to be 0.  

> 192.168.0.116-7E217B: [7] Update Characteristics
192.168.0.116-7E217B: Setter TH
192.168.0.116-7E217B: [6] Get Characteristics
192.168.0.116-7E217B: [6] Update Characteristics
192.168.0.116-7E217B: TH Off
192.168.0.116-7E217B: [7] Prepare
192.168.0.116-7E217B: [7] Update Characteristics
192.168.0.116-7E217B: Setter TH
192.168.0.116-7E217B: Setter TH
192.168.0.116-7E217B: [7] Prepare
192.168.0.116-7E217B: [7] Update Characteristics
192.168.0.116-7E217B: Setter TH
192.168.0.116-7E217B: Saving states
192.168.0.116-7E217B: TH Cooler
192.168.0.116-7E217B: Exec action 4
192.168.0.116-7E217B: HTTP/TCP Action 192.168.0.60:0
192.168.0.116-7E217B: ! DNS
192.168.0.116-7E217B: Saving states
192.168.0.116-7E217B: Read TH sensor

Hope this helps!

Thanks!

RavenSystem, 2020-07-06:
You must remove quotes from port key:
"p":8000

Try it.

bvermes17, 2020-07-06:
@RavenSystem 

That was it :)  After I removed the "", I could see the request in the log. The POST was still not working but using this I went back to what you originally recommended and crafted my own TCP RAW package. (Created the request in Postman and captured the packet with Wireshark) 

Now with RAW TCP it works perfectly :) Thanks for all the effort!

closed 2020-07-06 by bvermes17