RavenSystem/esp-homekit-devices #1030: Sonoff 4CH - Sprinkle icon & timer

halaszlaciii, 2020-07-01:
Hi there. I have a sonoff 4ch switch. I would like to switch my sprinkle in 4 zones. All I want is to be able to set each zone for a maximum of 1 hour. I want a smooth simple on and off switch and the icon could be in the shape of an irrigation. Can you give Json Config? :)

mat130780, 2020-07-02:
Hi,

this config should work. You only have to modifiy the "d" value to set the standard duration (in seconds):

`{"c":{"o":0,"l":13,"i":0,"f":10},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":20,"b":[{"g":0,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"t":20,"b":[{"g":9,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]},"t":20,"b":[{"g":10,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":20,"b":[{"g":14,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]}]}`

morphix9, 2020-07-15:
As far as I know, there is no support for irrigation in HAA yet (HomeKit supports it though), as already discussed in [this ticket](https://github.com/RavenSystem/esp-homekit-devices/issues/941). 
You can only set individual sprinklers, but they won't be connected into a zone. Unfortunately. I'm also waiting for this.

closed 2020-08-14 by stale[bot]