RavenSystem/esp-homekit-devices #1032: Can Sonoff be a relay?

danielssantos, 2020-07-01:
Dear, I am testing a new project, where through the sonoff I can open the gate. For this, I am using the sonoff basic connected to a relay, according to the attached images. However, reading some closed topics, I had the impression that this could be done, using GPIO and programming. Would it be possible?

Note: using the configuration shown in the photos, and the code below, I can open the gate normally.
Code: {"c": {"l": 13, "b": [{"g": 0, "t": 5}]}, "a": [{"0": {"r": [{"g": 12}]}, "1": {"r": [{"g": 12, "v": 1, "i": 1.5}]}, "b": [{" g ": 0}]," s ": 0," i ": 1.5}]}

[image]
[image]

i3laze, 2020-07-02:
Sure it can. The idea is to make dry contacts (isolate Sonoff relay from circuit logic).
Check this modding: https://www.instructables.com/id/Sonoff-Garage-Door-Opener/

Although I recently moved from Basic to Sonoff SV - it fits into Basic case and has 5-24V input (most garage boards have 24V output) and, most importantly, SV has 3 sensor inputs (closed, open, motion). Although I had to make dry contacts there too: https://youtu.be/FdfMTlo8hFM

closed 2020-07-04 by danielssantos

mestregroda, 2020-10-26:
Daniel vc conseguiu fazer funcionar o portao com contato seco?

mestregroda, 2020-10-26:
> Dear, I am testing a new project, where through the sonoff I can open the gate. For this, I am using the sonoff basic connected to a relay, according to the attached images. However, reading some [...]

Daniel vc conseguiu fazer funcionar o portao com contato seco?

danielssantos, 2020-10-26:
> Daniel vc conseguiu fazer funcionar o portao com contato seco?

Consegui, nesse modo que postei nas fotos: o sonoff + um relê. Aciono o comando no iPhone, o sonoff libera um pulso, que energiza o relê e fecha o contato. Fiz como se fosse uma “extensão” do interfone, para fazer funcionar. Consegui explicar?

mestregroda, 2020-10-26:
Cara vê se vc consegue me dar uma luz, to quebrando a cabeça e nao to conseguindo resolver.
Eu peguei um sonoff mini e fiz contato seco nele. Ai alterei o firmware pro tasmota. Funciona legal no homebridge. Consigo abrir e fechar o portao usando a siri (atraves da criacao de atalhos no ios) ou o app Casa, e ainda usando o controle remoto do portao tambem. Tudo funciona beleza, mas é uma gambiarra como switch e nao como garage door. Mas tava querendo era ver se consigo colocar o firmware da ravencore pra poder abrir o portao diretamente pelo carplay. Vi num site australiano que o cara conseguiu fazer isso, mas pra isso precisava ser garage door nativo no app casa. 
Como seria a fiacao disso no mini? Hoje ele esta ligado fase e neutro, energizando o sonoff e no contato seco eu liguei a botoeira e o gnd da placa da peccini 3030. Vc acha que funcionaria?

danielssantos, 2020-10-27:
> Cara vê se vc consegue me dar uma luz, to quebrando a cabeça e nao to conseguindo resolver.
> Eu peguei um sonoff mini e fiz contato seco nele. Ai alterei o firmware pro tasmota. Funciona legal no [...]

Vamos por partes: o meu também está como switch. Garage door tem funções demais para esse uso, na minha opinião. Tem um módulo lock mechanism q talvez te atenda melhor. Para abrir ele dispara o pulso e abre o portão. E tem a opção de colocar um sensor magnético para detectar se o portão está aberto ou fechado. Ainda sinaliza nos aparelhos se o portão foi aberto ou fechado. 
 Mas me tira uma dúvida: você fez tudo isso usando só o sonoff? Sem usar um relê Externo adicional?
 Sobre o CarPlay eu nunca testei, pela siri me atende bem. E no meu caso, eu abro o portão da portaria do prédio, não é um portão de garagem.

mestregroda, 2020-10-27:
Fiz usando só o Sonoff, o rele fica no contato seco. Vc saberia me orientar como funciona o JSON pra switch usando apenas como pulso (contato seco) e sem sensor magnetico?

danielssantos, 2020-10-27:
> Fiz usando só o Sonoff, o rele fica no contato seco. Vc saberia me orientar como funciona o JSON pra switch usando apenas como pulso (contato seco) e sem sensor magnetico?

Experimenta esse aqui:

Funcionando 100%, com tempo de 0,5 segundo, e sem interruptor externo. Gerado no https://glumb.github.io/haa-configurator/

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1,"i":0.5}]},"b":[{"g":0}],"s":0,"i":0.5}]}

OBS.: esta com tempo de 0,5 segundos, significa que ele dá o "pulso" por meio segundo. No meu caso, é mais q suficiente para abrir a fechadura elétrica do portão. Se precisar de mais tempo, é só mudar nas tags "i":0.5 .
Se puder, coloca uma foto do sonoff ligado no contato seco?

mestregroda, 2020-10-27:
Valeu! Teu JSON funciona como no Tasmota.
Abaixo segue as fotos do sonoff no contato seco:
[image]
Tem bastante video no youtube sobre essa modificacao. Basicamente corta a trilha dos NEUTRO e da FASE e jumpeia uma fase no antigo neutro.
[image]
O contato seco do sonoff conectei na botoeira e no comum da placa do portao.

Nesse setup: consegui fazer o sonoff funcionar como garage door. Funciona tudo que diz respeito a parte do homekit: Siri, Aparece o botao no Carplay, abre e fecha pelo app. Mas o que nao consigo é fazer funcionar o controle remoto. So falta isso pra ficar perfeito. Se vc puder ter uma ideia do que posso fazer me ajudaria muito. O JSON que to usando é esse:
{
  "c": {
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "v": 0,
            "i": 0.5
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 0,
            "i": 0.5
          }
        ]
      },
      "t": 40,
      "s": 5,
      "d": 14,
      "b": [
        {
          "g": 0,
          "t": 5
        }
      ],
      "f0": [],
      "f1": []
    }
  ]
}

E a foto do mardito controle:
[image]

danielssantos, 2020-10-27:
> Tem bastante video no youtube sobre essa modificacao. Basicamente corta a trilha dos NEUTRO e da FASE e jumpeia uma fase no antigo neutro.

 Era isso q estava confuso (para mim). Nunca testei essa opção de cortar as trilhas, descobri a opção do relê externo e fui nela, para minha necessidade, atendeu bem. 
 Sobre o teu JSON, fiz a validação e parece estar tudo ok, mas não sei como te ajudar, já que nunca mexi em acessório garagem door. Mas te recomendo os dois links abaixo, podem ser úteis:

https://jsonlint.com - É um validador. Copia e cola teu código, se tiver erro ele vai acusar. Poupa bastante tempo.

https://glumb.github.io/haa-configurator/ - É um gerador de JSON, você pode começar do zero, selecionando o tipo de acessório e quais funções você quer. E também do modo reverso: cola o código pronto e mostra o q tem nele, pode ser útil pra você entender o q cada tag faz e ajustar para sua necessidade.

mestregroda, 2020-10-27:
> > Tem bastante video no youtube sobre essa modificacao. Basicamente corta a trilha dos NEUTRO e da FASE e jumpeia uma fase no antigo neutro.
> 
> Era isso q estava confuso (para mim). Nunca testei essa opção de cortar as trilhas, descobri a opção do relê externo e fui nela, para minha necessidade, atendeu bem.
> Sobre o teu JSON, fiz a validação e parece estar tudo ok, mas não sei como te ajudar, já que nunca mexi em acessório garagem door. Mas te recomendo os dois links abaixo, podem ser úteis:
> 
> https://jsonlint.com - É um validador. Copia e cola teu código, se tiver erro ele vai acusar. Poupa bastante tempo.
> 
> https://glumb.github.io/haa-configurator/ - É um gerador de JSON, você pode começar do zero, selecionando o tipo de acessório e quais funções você quer. E também do modo reverso: cola o código pronto e mostra o q tem nele, pode ser útil pra você entender o q cada tag faz e ajustar para sua necessidade.
> 
> Obs: eu li de novo o q você escreveu e não entendi uma parte: você diz q não consegue fazer funcionar o controle remoto. Então você fez o sonoff funcionar, consegue operar ele pelo iPhone, mas o controle remoto do portão (chaveiro preto com botões azuis) deixou de funcionar, é isso?

Isso so falta fazer funcionar o botao azul do controle remoto. O botao branco no sonoff funciona direitinho, tanto no garage door quanto como switch. Valeu pelos links!

danielssantos, 2020-10-27:
> Isso so falta fazer funcionar o botao azul do controle remoto. O botao branco no sonoff funciona direitinho, tanto no garage door quanto como switch. Valeu pelos links!

E sem o sonoff, o controle funciona? O controle funcionava antes?

mestregroda, 2020-10-27:
> > Isso so falta fazer funcionar o botao azul do controle remoto. O botao branco no sonoff funciona direitinho, tanto no garage door quanto como switch. Valeu pelos links!
> 
> E sem o sonoff, o controle funciona? O controle funcionava antes?

Sim tudo funcionava beleza. Se eu mudar pro teu Json, funciona na hora, se mudar pro meu Json de garage door, ele funciona ate o momento que resolvo usar o homekit para o portao (seja clicando no botao do portao, seja via Siri ou seja via Carplay). Depois que ativa o sonoff via homekit o botao do controle remoto nao tem mais efeito.