RavenSystem/esp-homekit-devices #1034: Looking for json for LG TV remote

borissov, 2020-07-04:
Hello, 
I am looking for json config for LG TV remote. I use NodeMCU with BC547 + TSAL6100 connected on GPIO4 and is work with my script, but now I want to use it with HAA. 

Hire is all codes of LG remote:
https://gitlab.com/snippets/1690600

With this function my script works:
irsend.sendNEC(request.toInt(), 32);

I think the protocol is NEC. 
I tested with Panasonic example in wiki and led works. Thanks!

closed 2020-07-11 by borissov

Rahulsharma0810, 2023-09-25:
Able to figure this out ?

borissov, 2023-09-26:
Nope, I modified it by adding a DHT11 and turned it into an air conditioning controller with a custom script based on: https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266 .