RavenSystem/esp-homekit-devices #1036: sonoff sv lock mechanism

dosha25, 2020-07-07:
Hi. I add sonoff sv with templates lock mechanism. In homekit I see it. But relay don’t give 12v on 3s, what I do wrong?

hillaliy, 2020-07-07:
Sonoff SV - Loc mechanism with contact sensor

{"c":{"l":13,"i":0,"b":[{"g":0,"t":1}]},"a":[{"t":4,"i":3,"1":{"r":[{"g":12}]},"0":{"r":[{"g":12,"v":1,"i":3}]},"b":[{"g":0,"i":3}]},{"t":5,"f0":[{"g":5,"t":0}],"f1":[{"g":5}]}]}

dosha25, 2020-07-07:
> Sonoff SV - Lock mechanism with contact sensor
> 
> {"c":{"l":13,"i":0,"b":[{"g":0,"t":1}]},"a":[{"t":4,"i":3,"1":{"r":[{"g":12}]},"0":{"r":[{"g":12,"v":1,"i":3}]},"b":[{"g":0,"i":3}]},{"t":5,"f0":[{"g":5,"t":0}],"f1":[{"g":5}]}]}

Thanks work. Maybe you have templates Sonoff SV - Lock mechanism without contact sensor?

hillaliy, 2020-07-08:
"c":{"l":13,"i":0,"b":[{"g":0,"t":1}]},"a":[{"t":4,"i":3,"1":{"r":[{"g":12}]},"0":{"r":[{"g":12,"v":1,"i":3}]},"b":[{"g":0,"i":3}]} ]}

karnas99, 2020-08-01:
> Sonoff SV - Lock mechanism with contact sensor
> 
> {"c":{"l":13,"i":0,"b":[{"g":0,"t":1}]},"a":[{"t":4,"i":3,"1":{"r":[{"g":12}]},"0":{"r":[{"g":12,"v":1,"i":3}]},"b":[{"g":0,"i":3}]},{"t":5,"f0":[{"g":5,"t":0}],"f1":[{"g":5}]}]}

Hi,

can I ask you how to connect contact sensor? I just tried GIO5 pins S and G and I burned it. 

Thanks

karnas99, 2020-08-04:
> you need this sensor:
> https://he.aliexpress.com/item/4000138461602.html?spm=a2g0s.9042311.0.0.27424c4dssws1U
> connect one wire to gpis 5 and the other to ground

I have mechanic sensor integrated in the door lock but I believe it works pretty same. Have no idea why it was burned. Now I am waiting for new sonoff to be delivered

hillaliy, 2020-08-05:
the GPIO work on 3.3v
maybe your build in sensor work on 12v

karnas99, 2020-08-05:
> the GPIO work on 3.3v
> maybe your build in sensor work on 12v

build sensor is just switch it does't it does not produce any electric. However input voltage is 12V

hillaliy, 2020-08-05:
that what I mean. the input is 12v and the output from the SV is 3.3. because of that it's burn.

karnas99, 2020-08-06:
> that what I mean. the input is 12v and the output from the SV is 3.3. because of that it's burn.

Alright so I have connected 12V power input which is giving 12V output and unlock door lock. With this circumstances is there any way how to use sensor - switch on  GPIO 5 and not to burn it?

hillaliy, 2020-08-07:
yes use 3.3 contact sensor

closed 2020-09-06 by stale[bot]