RavenSystem/esp-homekit-devices #1038: Controlling a gas fireplace with a 0-3V signal

tokamac, 2020-07-08:
I'd like to control a gas fireplace with HomeKit. The model I want to control is made by Dutch company DRU. 

ON, OFF and anything "dimmed" in-between (flame height) is controllable with a 0-3VDC low-current input signal. Actually, it is even 0-2V as anything equal or above 1.98V and up to 3V is an OFF command.

The "dimming" function (flame height) is any voltage going from 0V (0%) up to 1.47V (100%).

If the voltage cannot reach exactly 0V, no matter it will just tell the flame height to be low enough, by a few percent.

As the power (both voltage and current) of the DC signal is very low (we are not driving LEDs here) I think that a very simple digipot used as a variable voltage divider would do the trick.

Is there a solution to make this work with ESP8266 and HAA? I was thinking of making a circuit with a wirelessly controlled 100K digipot acting as a varying voltage divider (?) But maybe some commercial devices already exist on the market doing exactly this?

Alternately, I was initially thinking about a 0-10V dimmer with its output stepped down to 0-3V (as suggested in the DRU notice) but there is no HomeKit-compatible 0-10V controller on the market today, nor any Wi-Fi one.

[image]
[image]
[image]

Bene2103, 2020-07-10:
First of all, i dont know your setup nor any other gas fire places, but be safe and dont bungle around when your not knowing what you are doing! As an industrial i know litte about explosion proof things and thats an huge theme.

Do it at your own risk.

So about your questions, i think it would be possible to let haa act like an Led or in this term as an LightBulb. Haa will give out an PWM signal (you can set the frequency in config) which is between 0 and 3.3V or at least near that voltages.
If you set up an Low-Pass filter on the output you can convert this PWM into an analog voltage. [Like described here ](https://www.allaboutcircuits.com/technical-articles/low-pass-filter-a-pwm-signal-into-an-analog-voltage/).
To get the analog voltages you like, you can always set an voltage divider on the output of the Low-Pass filter. So 100% in Homekit would be 1,47V. The only thing to consider is, is 0% (0V) the same as OFF (>1,98V)? This could be a problem, but i think it would be possible to use two digital outputs one for PWM and the other on for the OFF which are switched inverted so PWM=0% would switch the other switch on and you have 3V on your Fireplace as the OFF signal

tokamac, 2020-07-10:
Thanks for the answer. About the security issue: I perfectly understand that you have to put these safe forewords. To explain the context a bit more on my side, first the electric box is separated from the fireplace and located in a closet at some distance. Second, the signal involved is low power, low voltage. Third, the fireplace box is designed for such controlling signal and the only thing that a wrong voltage (> 3V) could do would be to burn the integrated home automation controller of the fireplace (making it ineffective) but with no issue on the burner itself. So yes, I write it: I'll do it at my own risk. But obviously I'll be particularly attentive to all security procedures relative to electric, heat, gas and flame hazards.

OK for the part involving a low-pass filter. Otherwise, would [this little cheap DAC](https://www.aliexpress.com/item/33005505745.html) do the same thing? I wonder if you have to program it… would it react on its own to changes in the PWM signal output by HAA, i.e converting a 2V PWM signal output in a 2V analog signal "naturally" without programming, as for the low-pass filter?)

I agree that at first it seems 0V seems to have the same effect as the voltage range [1.98V-3.00V] i.e. an OFF command, like for lightbulbs in HomeKit where 0% = OFF. But in fact, in the case of this fireplace, I think OV = 0% is actually still an ON command, as it triggers the PILOT mode of the burner (a very little flame still active, like the pilot flame in some gas boilers). 

This pilot flame is not mandatory in everyday life, to say the least (it is in fact completely useless). So the real-world commands would be :
1% - 100 % (dimming function): from a few millivolts to 1.5V
0% or OFF command: 2V (or anything between 2V and 3V)

Do you kn ow what kind of commercial device based on the ESP8266 could do the trick? Sonoff SV?

Bene2103, 2020-07-10:
I dont think this DAC would work cause of the I2C interface which (i think) isnt supported by HAA.

Theroretically should a single ESP-Board already work for your needs, like an NODEMCU. Besides the PWM to analog conversion everything is fine with this device. But you should consider an voltage divider so you can set the voltages the way you dont damage the fireplaces electronics.

for Example
GPIO1 -> PWM Signal -> Low-Pass -> voltage divider -> signal input
GPIO2 -> voltage divider -> OFF signal 

The you connect both of the GPIO's to the input of your fireplace and set an Homekit automation like this:
IF PWM = 0 THEN GPIO2 ON
IF PWM >0 THEN GPIO2 OFF

I dont know exactly but i think this would be also possible inside HAA itself, even i dont know how to do it

I'm not an electronics expert but i think this could work, maybe a shottkey diode for reverse voltage protection could be added.

Where are the electronics experts? :-)

tokamac, 2020-07-10:
Many thanks. Yes I know how a voltage divider works using resistors, and how it is mandatory in this case to prevent anything over 3V to prevent electronics from frying (or 2V max in practice for this specific control). The addition of a one-way diode is a good advice too, even from a non-electronics expert ;)

I don't know if your example with the IF condition could be made into Home or Shortcuts apps. I think it has to be done programming at the HAA level, perhaps through the [Wildcard Actions](https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#Wildcard-Actions) or other action types described in the wiki.

However, like you I'm rather stuck on how to implement which code. Help from HAA devs would be much appreciated.

PS: When I asked for any "commercial" device, I didn't speak of any plug & play smart device doing all this out of the box (I know it doesn't exist as I first searched the internet for it) but the example of NodeMCU you cited as a solution seems perfect.

tokamac, 2020-07-10:
I may have the solution, not with one but two voltage dividers as shown below. I took the pins of a Sonoff SV, which I have already at home in front of me (any advice against a Sonoff SV vs NodeMCU in this case?)

The first voltage divider is made of 3K + 1.3K resistors connected to GPIO4, while the second voltage divider is made of 3K + 2K resistors connected to GPIO14. With this configuration I get two different maximum voltage values, one at 1.5V max (corresponding to 100% dimming) and the other at 2V for switch off.

[image]

I didn't represent the low-pass filter nor the Schottky diode in the diagram, but you get the idea.

The Sonoff SV board would be configured with HAA as a lightbulb and powered with a small 5VDC power adapter.
When dimming is set between 1% and 100%, GPIO14 is OFF; GPIO4 is ON and outputs a variable PWM voltage from a few millivolts min value (1% dimming) to about 1.5V max value (100% dimming) according to the dimming level asked via HomeKit, which sets the flame height of the fireplace. 
When dimming is set to 0%, or the lightbulb is switched OFF, GPIO14 is ON and outputs contant 2V, which switches the fireplace burner OFF.

Do you think this hardware setup could work? Besides the HAA code I still do not have.

Bene2103, 2020-07-10:
You got it! That was what I meant, but be aware of the voltages from the ESP which is not necessary 5VDC because the ESP works on 3.3V and the GPIO's are also on 3.3V. If you want to be sure get a datasheet from the sonoff or measure the voltages on the GPIO's. But I think this should work, I'd give It a try.

The JSON for HAA should not be too hard, if I have some time I take a look into it but I actually don't know anything about wildcards or actions.

Maybe someone else could help here?

tokamac, 2020-07-10:
Yes, of course is the voltage on GPIO pins is 3.3V while the board is powered with 5VC, I would choose the resistors accordingly. 2K + 1.7K and 3.2K for example. Better to check the exact voltage output first.

closed 2020-08-09 by stale[bot]

tokamac, 2020-10-19:
It seems that Sonoff SV cannot be configured as a lightbulb (with PWM dimming capability). After saving smallest possible JSON in the captive URL, the accessory can't be added to HomeKit (error "device already added" even with HK reset). 
JSON for lightbulb is `{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"w":14}]}`
I also tested with lengthier options in JSON to no avail.
When configuring as a basic switch again, device can be added with no issue.