RavenSystem/esp-homekit-devices #1039: not an issue i have an updated example water valve

Hucksleyy91, 2020-07-08:
not an issue i have an updated example json for a water valve type as the one in the current example does not work :)
was not sure how to add this

{
  "c": {
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "v": 0
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ]
      },
      "t": 20,
      "b": [
        {
          "g": 0,
          "t": 1
        }
      ],
      "s": 0,
      "w": 1,
      "d": 80,
      "f1": [],
      "f0": []
    }
  ]
}

WizBangCrash, 2020-07-09:
Hi @Hucksleyy91,
What type of water valve was this configuration for?
I'm in the process of updating the wiki and would like to include your example.

Hucksleyy91, 2020-07-09:
This config is for a sonoff sv 
I would change "d": 80 to "d": 3600 for an example as it was personal preference

WizBangCrash, 2020-07-12:
Thanks. I've added it into the water valve example now.

closed 2020-07-14 by WizBangCrash