RavenSystem/esp-homekit-devices #1041: Magichome rgb

dosha25, 2020-07-08:
Hi. I search templates with flash led mode. Please help

hire-vladimir, 2020-07-25:
I assume this is for RGB led strip? If its one of these https://tasmota.github.io/docs/devices/MagicHome-with-ESP8285/

i've had good luck with below config.

```
{
  "c": {
    "q": 850,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "t": 30,
      "r": 5,
      "g": 12,
      "v": 13,
      "w": 15,
      "b": [
        {
          "g": 1
        }
      ]
    }
  ]
}
```

dosha25, 2020-07-26:
> I assume this is for RGB led strip? If its one of these https://tasmota.github.io/docs/devices/MagicHome-with-ESP8285/
>
> i've had good luck with below config.
>
> ```
> {
>   "c": {
>     "q": 850,
>     "b": [...]

Thanks my friend. But I search templates with flash effects:
[image]

vijaydembla1, 2020-08-01:
Hey
Any luck with the template search and config?

dosha25, 2020-08-01:
> Hey
> Any luck with the template search and config?
Unlucky. I search templates and this time

closed 2020-08-31 by stale[bot]

than-sad, 2023-03-09:
Hi, I found this page referring to how to define the effects, but I don't know how to made it in HAA and I can't see any references to effects in Wiki.

Can anyone give us some tips on how to do it?

Best regards.