RavenSystem/esp-homekit-devices #1043: Control devices trought Siri on iOS 14.b2 don't work

Chuck3CZ, 2020-07-12:
I dont know if it is issue of iOS 14.b2 or HAA/RC, but not certified **EIIE rev-1 works** on this newer iOS, trought Siri, but **HAA 0.9.2-2.5.1 doesn't**

camilogrifi, 2020-07-18:
I have this problem!!! :(

PEA71, 2020-07-18:
Same issue for me. Let's see if this issue persist over next beta (probably next week).

PEA71, 2020-07-23:
Issue not resolved with beta 3...
If I try to do a Save with the "Controller" app, it seems that it cannot read the information of HAA devices. But not sure if it's linked. [image]

peros550, 2020-07-24:
hi guys, I can confirm the issue on beta 3. Also, I'm aware that similar issues exist for accessories brought to Homekit through Home Assistant hub. 
Interestingly, I do still have a sonoff mini which I never bothered to open it and install HAA. I'm using it though this homebridge plugin. This one seems to be controlled by Siri with no issues on ios14b3.

pkordal, 2020-07-25:
Same here. It works well for original devices like Hue or Aqara or even homebridge but not for devices with HAA.

peros550, 2020-07-25:
Question is : what HomeBridge does and Siri works with it

Chuck3CZ, 2020-07-27:
Homebridge Works, TimeThinker socket works well too (EIIE), only on HAA Siri says - not found.

PEA71, 2020-08-06:
Issue is still present with iOS 14 beta 4. 
I have an older iPad with iOS 12.4.8 and everything is working as normal. 
There probably a little change under the hood in iOS 14 that cause this issue...

angyalZK, 2020-08-06:
Issue still on iOS 13.6. Sometimes Siri can switch on the light, but can't switch off. Or says "done" but does nothing.

Murdrock, 2020-08-18:
Anyone tried beta 5 !?

Chuck3CZ, 2020-08-18:
Reported to Apple with Siri debug profile, they asked for it, and now it works, closing issue😍🙌🥰✅✅✅

closed 2020-08-18 by Chuck3CZ

PEA71, 2020-08-26:
I confirm! All work as expected with Siri now! Thanks @Chuck3CZ !
👍🍻