RavenSystem/esp-homekit-devices #1045: Going from HAA to Tasmota with OTA

serkanyilmaz, 2020-07-13:
Hello there, is there a way to go to Tasmota from HAA by using OTA, I do not want to open the device as it is in a difficult place to access. Thanks

ptath, 2020-08-01:
Just upload haa from tasmota web interface and it will work fine, just tried from 6.5.0 — works perfectly

prichard84, 2020-08-03:
> Just upload haa from tasmota web interface and it will work fine, just tried from 6.5.0 — works perfectly

I don't think you understand, our friend wants to go from HAA to Tasmota, I have this same problem in a lightbulb, I'm giving it to my mom, she has an echo dot, personally I think it should be through the web interface as we do at Tasmota.

AntlerDM, 2020-08-12:
I am also hoping to do this with one of my RGB lightbulbs. All of the other tuya devices I have are getting HAA installed as fast as I am able, but I need to downgrade a few HAA lightbulbs and tuya-convert seems to be disabled (for now) with the new tuya firmwares.

+1

closed 2020-09-11 by stale[bot]

maslianok, 2020-11-17:
yes, it would be great if HAA will provide such options.

It's very easy to update from Tasmota to HAA, but I'm not sure how to do the opposite.

P.S. I have Sonoff Mini with HAA and I'd like to change the firmware to Tasmota

retrography, 2021-09-29:
2021 and counting. Yes that would be very useful.

Fudalii, 2022-05-30:
I also need it.

panache67, 2024-12-09:
Given it’s almost end of 2024. Has there been any movement since?

daryelv, 2025-01-22:
Given that the developer has no interest in you leaving his ecosystem, he's not going to provide a way to do it.  Unfortunately this means you either 

1) buy all new devices - this would run me around $400-$500 for the devices plus all the time to replace the firmware on each new device with Tasmota, and finally all the time to physically replace the HAA device with the new Tasmota flashed one.  I'd estimate around 20-32 hours of labor to do this with the amount of devices and the locations they're in.

2) refresh existing devices - since you need to use hardware to reflash, this involves uninstalling the device, re-flashing, re-installing.  This isn't even possible on some devices that don't come apart (small outlets for example).  I'd estimate the hours to be around 48-56 total for all the devices in my setup.

3) a developer here embraces "Open Software" standards or provides a way to flash Tasmota over HAA.  Even if it was an old version of Tasmota that was 'blessed' with the signed keys that are required for flashing, that would at least let us reflash something like a ceiling fan control (I have 2) which are very difficult to get to.  

Considering this disclaimer was not added until July 14, 2022, I feel that a tool to allow flashing HAA back to Tasmota should be made available to anyone that has been stuck using HAA since before July 14, 2022. 

[image]