RavenSystem/esp-homekit-devices #1047: Does RavenSystem have a sleep timeout or something?

jlg89, 2020-07-14:
I loaded RavenSystem on a Sonoff SV and configured it per the wiki as a gate opener with contact sensor, using the Garage Door sample code. Added it to Homekit, everything works. But the HAA device goes "no response" in Homekit after a period of non-use. I'm looking at the General Configuration options, and thinking maybe tweaking the ping or mDNS parameters might help. Am I on the right track? Any guidance is appreciated.

komarofleo, 2020-07-14:
Can you share pls the json code? I am looking for the code to use with hall sensors. Is it possible?

jlg89, 2020-07-14:
Look in the wiki under the Garage Door configuration...that's the json code I used. If your hall sensor is doing the same thing as a magnetic switch sensor (i.e. indicating open/close state by opening/closing a wiring loop), it should work. If it's a solid-state switch, of course polarity will matter.

komarofleo, 2020-07-14:
Thanx for respond - can you tell me which GPIO should I connect to the sensor? I think magnetic is the same )

jlg89, 2020-07-14:
See https://www.itead.cc/wiki/Sonoff_SV 

It's the pins marked "IO14" using the "S" and "G" pins (Signal & Ground).

[image]

jlg89, 2020-07-15:
I thought I had fixed it, but no. The SV still goes offline in Homekit, but is still pingable on the network. Bouncing power brings it back online. So I'm looking for some way to make the Sonoff SV give a "heartbeat," or something like that, to Homekit on a regular basis.

RavenSystem, 2020-07-15:
You have a bad network mDNS environment. Use "ttl":60 config option as workaround.

jlg89, 2020-07-16:
You got me on the right track. I updated the config with the "ttl":60 parameter, and that didn't work. But then I started thinking about the multicast issue. This gate opener is 1800 ft. from the house, and to get network out there I'm using hardware that bridges Ethernet/PoE across a coax cable to a low-power AP and the Sonoff SV. I'll bet those things don't pass multicast. So, it looks like what I need to do is use the avahi daemon on my HOOBS server to advertise the remote services of the Sonoff SV. Can you give me a clue as to what mDNS services need to be advertised? Or is just the hostname enough (i.e. "HAA-xxxx.local")?

jlg89, 2020-07-16:
So far, it's looking something like this:

`<?xml version="1.0" standalone='no'?>
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<service-group>
  <name>HAA-448ED2._hap._tcp.local</name>
  <service>
    <type>_hap._tcp</type>
    <port>5556</port>
    <host-name>HAA-448ED2.local</host-name>
    <txt-record>md=RavenSystem HAA</txt-record>
    <txt-record>pv=1.1</txt-record>
    <txt-record>id=48:59:97:98:22:29</txt-record>
    <txt-record>c#=8454</txt-record>
    <txt-record>s#=1</txt-record>
    <txt-record>ff=0</txt-record>
    <txt-record>sf=0</txt-record>
    <txt-record>ci=1</txt-record>
    <txt-record>sh=utuKeg==</txt-record>
  </service>
</service-group>`

closed 2020-07-17 by jlg89