RavenSystem/esp-homekit-devices #1048: HTTP PUT command - &param being converted to ¶ (bug?)

rjblake, 2020-07-15:
Have tried to include an HTTP PUT command into the JSON, but it does not get accepted as updates part of the command string from "&param" to a paragraph marker "¶m". Can't seem to find a way around this, so perhaps a bug in the HTTP code?

**Relevant line of JSON:**
`"f2":[{"g":12,"t":0},{"h":[{"h":"83.84.217.76", "p":8084, "u":"json.htm?type=command&param=switchlight&idx=613&switchcmd=On"}]}]`

**Gets saved as:**
`"f2":[{"g":12,"t":0},{"h":[{"h":"83.84.217.76", "p":8084, "u":"json.htm?type=command¶m=switchlight&idx=613&switchcmd=On"}]}]`

rjblake, 2020-08-10:
Any update on this issue?

rjblake, 2020-09-04:
@RavenSystem - has this been fixed in current version? Can't seem to find any reference to it in the latest code, but could have missed it.

rjblake, 2020-09-30:
@RavenSystem - is this on the backlog to fix at all?

RavenSystem, 2020-10-12:
JSON is saved fine. Special character appears only into html text area.

rjblake, 2020-11-02:
> JSON is saved fine. Special character appears only into html text area.

Understand that this is correct first time around. If you connect again to edit/modify JSON, you need to make sure to replace all instances of "¶" with &para or else incorrect JSON is saved.

RavenSystem, 2020-11-09:
> Understand that this is correct first time around. If you connect again to edit/modify JSON, you need to make sure to replace all instances of "¶" with &para or else incorrect JSON is saved.

I must test it.

RavenSystem, 2021-06-04:
Process of replacing that rare character can need so much RAM in some scenarios. I will no fix it.

As workaround, make sure to replace all instances of "¶" with &para or else incorrect JSON is saved.

closed 2021-06-04 by RavenSystem