RavenSystem/esp-homekit-devices #1049: Will HAA work on Shelly i3?

gijsstegehuis, 2020-07-15:
A Shelly i3 would make a perfect Homekit stateless button.
Can it be done using the current firmware?

It has three channels. Would it be possible to make it show up as 3 stateless buttons in Homekit?

RavenSystem, 2020-07-15:
You can install HAA to Shelly i3, and you could create 3 stateless buttons in HomeKit, or 2 buttons and one movement sensor, or which you want. You have 3 digital binary inputs.

gijsstegehuis, 2020-07-21:
Just received the Shelly i3's. Will test asap.
Do you happen to know the GPIO numbers for the inputs? it's not on their official website.

RavenSystem, 2020-07-21:
You can follow circuit tracks, or you can setup a simple homekit switch and attach a GPIO number to it, and then change GPIO number and test it. You can create a bunch of switches too.

lociii, 2020-07-24:
[image]
[image]

Here's the inner workings of the i3 with circuit tracks from input to the ESP

gijsstegehuis, 2020-07-24:
@lociii awesome! danke

lociii, 2020-07-24:
Don't expect too much. I couldn't get it running with ESPHome as the inputs are floating and I didn't find any way to make them behave correctly.

gijsstegehuis, 2020-07-24:
I am not into the hardware side of things, but isn't this wat the pull-up resistor is for?
There's a config setting for it in HAA.

lociii, 2020-07-24:
Usually yes, but with pullup enabled, there is no signal at all.
I mean, try your luck. I didn't have some :)

gijsstegehuis, 2020-07-24:
Thx. I will see if i can get it working.

gijsstegehuis, 2020-07-25:
@lociii what sort of connector or cables are you using to flash the i3?
I soldered cables to a male/male header pin with 1,27mm pitch, but i can't get a decent connection to the ftdi adapter.

are there any standardized connectors around?

lociii, 2020-07-25:
I used a self made 1.27 pitch connector too. You have to make sure they do in deep enough and are all pretty much the same length.

AndreWillems, 2020-07-26:
just soldered some wires to the (small) connector. see pin-out on the picture. I flashed it the usual way like all other Sonoffs and Shelly's and it works perfect. I'm not using Tasmota or anything, build my own MQTT code for such IoT devices. 

[image]

lociii, 2020-07-27:
@AndreWillems was there anything special you had to change on your code to get the inputs working? Is it publicly available?

AndreWillems, 2020-07-27:
> @AndreWillems was there anything special you had to change on your code to get the inputs working? Is it publicly available?

@lociii, no nothing special. obviously changes the inputs based on the 3 switches from this Shelly device. Flashed it serial once and after I use OTA. drop me a mail if you want my source code.

gijsstegehuis, 2020-07-27:
@AndreWillems can you share the json configuration you used?

AndreWillems, 2020-07-27:
> @AndreWillems can you share the json configuration you used?

sorry @gijsstegehuis, I'm not using HAA or Tasmota. just wrote my own code once for Sonoff/Shelly and used this for Shelly I3 as well. no JSON config files.... but hopefully the picture with IO's can help you/others to make the appropriate JSON file ?

closed 2020-08-26 by stale[bot]

craigueh, 2020-09-10:
Hi AndreWilliams

Could you provide your code to flash to the i3?

I tried esphome but can’t get the floating inputs to behave correctly.

martinjuhasz, 2020-09-24:
I managed to get it working. You need to add a delayed_on_off. See my sample config here:
https://github.com/esphome/issues/issues/1459

gijsstegehuis, 2020-11-10:
For anyone stumbling upon this issue:
https://github.com/mongoose-os-apps/shelly-homekit just released version 2.5.0 which has support for Shelly i3

pilot1981, 2021-01-08:
anyone is able to use shelly i3 with HAA? can you share JSON?

RavenSystem, 2021-02-02:
`{"c":{"l":0,"b":[{"g":2,"t":5}]},"a":[{"t":3,"f0":[{"g":14,"p":0,"m":1}],"f1":[{"g":14,"t":2}],"f2":[{"g":14,"t":3}]},{"t":3,"f0":[{"g":12,"p":0,"m":1}],"f1":[{"g":12,"t":2}],"f2":[{"g":12,"t":3}]},{"t":3,"f0":[{"g":13,"p":0,"m":1}],"f1":[{"g":13,"t":2}],"f2":[{"g":13,"t":3}]},{"t":22,"n":5}]}`

sonicpixel2020, 2022-02-09:
Hey. Thanks for this Config. I would like to have Sensor to see if the switch is on or off. Is that also possible? I could not get this working. Can you send a Config for this? Thank you.