RavenSystem/esp-homekit-devices #1050: Firmware Shows 2.5.1 with an available update to 3.4.0

Fannangir, 2020-07-16:
Hi,
Not sure whether this is an issue or I am missing something.

When I log into HAA, I can see firmware 2.5.1.
It says firmware 3.2.0 is available.
I have ticked the option to update and enabled OTA update (without entering any custom repo).
Entering into HAA again, I can see that firmware does not update.
What am I missing?

Thanks.

tonysprenk, 2020-07-16:
The version that is shown in setup modes is the version of HAA OTA. HAA OTA is the piece of the firmware that updates HAA main (latest version 2.5.1).

Fannangir, 2020-07-16:
Oh, understood.
So how can I see that the device has received 3.2.0 update? Or what would be the current version of the main firmware?
Thanks.

tonysprenk, 2020-07-16:
When you select auto update and save it updates to latest for both. You can see current version of HAA in the info of the device in the home app.

Fannangir, 2020-07-16:
So I am probably missing something.
This is what I see in Home App:
[image]
It says 2.5.1.
I have ticked OTA and Update now.
I have tried to modify JSON string as well (to verify that data is actually saved).
Thanks.

tonysprenk, 2020-07-16:
Yup 2.5.1 is the latest version.

Fannangir, 2020-07-16:
So back to my initial question.
How do I know that the device has received 3.2.0 main firmware update?

tonysprenk, 2020-07-16:
Main firmware version (latest: 2.5.1) number is what you see in the home app. OTA firmware version (latest 3.2.0) number is what you see in the setup mode screen. Both are on latest in your case.

Fannangir, 2020-07-16:
OK, understood.
I was sure that the version I see in the brackets is the available update.
Thank you very much for clearing it up to me.
Thanks.

closed 2020-07-17 by Fannangir