RavenSystem/esp-homekit-devices #1051: Config for Contact Sensor

vijaydembla1, 2020-07-17:
Hi,

I love this and have been playing a lot lately. The place that I'm stuck at is the door sensor. It's connected to GPIO 5, that's the only accessory that is not working out of the whole in the config below. Am I missing something?

Any help will be appreciated

{
	"c": {
		"l": 13,
		"b": [{
			"g": 0,
			"t": 5
		}]
	},
	"a": [{
    "t": 1,
    "s": 0,
    "0": { "r": [{ "g": 1, "v": 1 }] },
    "1": { "r": [{ "g": 1, "v": 0 }] }
    },
	{
    "t": 1,
    "s": 0,
    "0": { "r": [{ "g": 3, "v": 1 }] },
    "1": { "r": [{ "g": 3, "v": 0 }] }
    },
	{
    "t": 1,
    "s": 0,
    "0": { "r": [{ "g": 15, "v": 1 }] },
    "1": { "r": [{ "g": 15, "v": 0 }] }
    },
	{
    "t": 1,
    "s": 0,
    "0": { "r": [{ "g": 13, "v": 1 }] },
    "1": { "r": [{ "g": 13, "v": 0 }] }
    },
	{
    "t":30,
    "w":12,
	"s":0,
    "b":[{"g":0}]
  } ,
	{
	"t": 12,
	"i": 5,
	"f1": [{ "g": 4 }]
	},
	{
		**"t": 5,
		 "f1": [{ "g": 5 }]**
	},{
		"t": 40,
		"d": 5,
		"s":5,
		"0": {"r": [{"g": 1,"v": 0,"i": 0.5}]},
		"1": {"a": 0},
		"2": {"r": [{"g": 1,"i": 1.5}, {"g": 1,"v": 0,"i": 0.5}, {"g": 1,"v": 0,"i": 2}]},
		"3": {"a": 2},
		"f3": [{"g": 14,"t": 0}],
		"f4": [{"g": 14,"t": 1}]
	}]
}

RavenSystem, 2020-07-17:
`{"t":5,"f0":[{"g":5,"t":0}],"f1":[{"g":5}]}`

vijaydembla1, 2020-07-18:
Great, it worked. Thanks again :-)

For some reason, the motion sensor is not working , I understand the "i" condition and don't want to use it just yet, however if i connect the PIR sensor to 3.3v pin, it is always low and if it connected to 5 v it is always high. Tried 2 different sensors, am i mission something. Your response is greatly appreciated.

"t": 12,
"i": 5,
"f1": [{ "g": 4 }]

Thanks

ZivAssor, 2020-08-17:
> {"t":5,"f0":[{"g":5,"t":0}],"f1":[{"g":5}]}

I have this door sensor: https://templates.blakadder.com/TY01.html
you think it's gonna work? or should I change something?
Thanks