RavenSystem/esp-homekit-devices #1054: OTA update from stock firmware

gijsstegehuis, 2020-07-21:
Just fried the flash/debug pins on my new Shelly i3.
...which made me wonder: can HAA be installed from stock firmware by [using the same trick they use here](https://github.com/mongoose-os-apps/shelly-homekit#updating-from-stock-firmware)?

pfandrade, 2020-08-16:
This is an excellent question. Would make flashing devices so much easier.

brymonster, 2020-08-24:
No kidding, especially for those devices that are stuck on lower HAA unable to update OTA. I still haven't had the time to reflash them manually!
But OTA initial install would be a removing a huge barrier to entry into the HAA system.

wayn3, 2020-09-19:
I think this can be used as an intermediate firmware to jump to HAA https://github.com/yaourdt/mgos-to-tasmota

_however I haven’t tried it_

RavenSystem, 2020-10-12:
https://github.com/yaourdt/mgos-to-tasmota

closed 2020-10-12 by RavenSystem