RavenSystem/esp-homekit-devices #1055: Setup Mode Accessory

jasongoldsmith, 2020-07-24:
Create a special accessory type that reboots the device into setup mode.

Suggest something like

"t": -1

It presents as a stateless button in HomeKit. Tapping the button will reboot the device into setup mode with no additional configuration.

Saves on characters in JSON and greatly simplifies initial setup and configuration flows.

Optionally add this as a checkbox in the web UI to enable the setup mode button.

ymyshko, 2020-07-25:
From Wiki

System Actions

System actions are defined by the "s" key array within an accessory object.

System actions were introduced in firmware version 0.8.7

Key	Value	Description
"a"	0	Reboot the device
1	Enter device setup mode
2	Search for OTA firmware update (only if firmware installed via OTA method)

jasongoldsmith, 2020-07-26:
@ymyshko Yes, that's another way to do it - but it takes a lot more JSON to implement

This JSON works too: 
{
  "a": [
    {"1":{"s":[{"a":1}]},"i":0.5}
  ]
}

But it isn't the point - it makes it overly complicated for a basic necessity for setting up a device that doesn't have a known working template

closed 2020-08-26 by stale[bot]