RavenSystem/esp-homekit-devices #1062: Request for magichome RGBWW

toshibochan, 2020-07-29:
I try magichome RGBW from exemple here and work good. But from this setup when I turn on white color the cold white(RGB) and the warm led turn on. But I have RGBWW and I want only the W be on when white color is on. Anyone how I can do that?

toshibochan, 2020-08-01:
Anyone can help?

toshibochan, 2020-08-01:
Hi @RavenSystem you can look this https://github.com/maccoylton/esp-homekit-rgbw-strip.  Maccoylton Magichome version have pure white opinion in he’s code. I was using he’s code but it keep loosing connection to homekit. So I start using yours today. But I love to have pure white if possible. Can you take look he’s version and try to and in yours magichome version. Please
Thank you for sharing amazing work with us.

closed 2020-08-31 by stale[bot]

kevinjohncutler, 2020-09-06:
Shoot, I'm also trying to figure this out.