RavenSystem/esp-homekit-devices #1067: The latest version is not added to homekit

exrector, 2020-08-02:
Something strange. I installed the latest firmware in two different ways (UART and from Tasmota). But device (magic hame rgb) doesn't appear for Homekit. Nothing helps.

toshibochan, 2020-08-02:
After you setup json and WiFi will take about 3 to 5 min to install all the thinks and show in the home app for pearing. Try to wait.

exrector, 2020-08-02:
> After you setup json and WiFi will take about 3 to 5 min to install all the thinks and show in the home app for pearing. Try to wait.

I know how this process goes.  I have several flashed devices.  I'm talking about the latest version.  device does not appear for HomeKit.  Even with the "wrong" json, the device should appear.  but no.  meanwhile the device is online and I can see it on my router

pawelosin, 2020-08-04:
The same issue.

hillaliy, 2020-08-05:
Same at me

tlagarto, 2020-08-08:
Same here... aparentely.
I flashed 3 news sonoff mini, I get to config and use a already validated json string but after saving it never shows on homekit.
I also noticed that the sonoff mini physical button does not respond. Is this something to do with the new fw?

Luistiri, 2020-08-08:
Same Problem... @exrector did you find a solution?

pawelosin, 2020-08-08:
https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_2.5.7

pawelosin, 2020-08-08:
Now everything works fine.

exrector, 2020-08-18:
> Теперь все работает нормально.

have you updated the fullhaaboot file? because the problem is still there. 
I used a brand new device. the experiment is clean.

exrector, 2020-08-18:
please help me.  the device is flashed.  device status online (in the router).  I went through a dozen json but the device does not appear to connect in the home application.

exrector, 2020-08-19:
I think I found the cause of the problem.  after reflashing, I connected the native power.  it doesn't work like that.  after the firmware, you need to turn on the power from the 3.3v flasher to configure HomeKit and only then turn on the native power.  you made a mistake somewhere in the code with gpios
Как всегда всё сам......блять

closed 2020-08-19 by exrector