RavenSystem/esp-homekit-devices #1069: CarKey

xrust83, 2020-08-05:
Did Carkey devices appear among the list of X
HAA devices? So that Siri can open/close the car with her voice.
[image]

tonysprenk, 2020-08-05:
Not sure what you mean by this. But as far as I understand CarKey is not a feature of HomeKit. It's a stand alone service that works through the wallet app. Kind of like ApplePay.

xrust83, 2020-08-05:
I think this is not difficult to implement by connecting an ordinary lock and calling it the name of the car.  But there is one nuance you need to install a wifi router in the car.  It is possible to create a hamster device that will work together with a phone without wi-fi directly.  Perhaps my question is deep, I beg your pardon.  but if for example sonoff SV (5v dc) flash it into haa Lock Mechanism (4 or 44) will it work directly with the phone without wi-fi?

tonysprenk, 2020-08-05:
The implementation wouldn't be that hard. Yes you need a WiFi network for HAA to work. The thing to think about is that most cars need the key to start, so if your going to take the keys anyway you might as wel unlock the car with the keys. One feature of CarKey is that you can also start the car with your iPhone.

tonysprenk, 2020-08-05:
Probably better to wait for someone to reverse engineer CarKey so that it can be implemented after market.

xrust83, 2020-08-05:
My car has a Start-Stop button and I don't need keys.  I can also open the car with a device from AliExpress.  But this device primitively sends a command via bluetooth.  And the problem is that I have to get my phone and open this app.  I don't need to press the open button if I am close to the car.  but getting the phone and opening the program is not convenient.  I would like to tell Siri open through your apple watch or airpods

xrust83, 2020-08-05:
[image]

tonysprenk, 2020-08-05:
Have a link to the device?

xrust83, 2020-08-05:
now I have this version
https://a.aliexpress.com/_dTa8XoW
to work, you need to connect to the bluetooth device, after which everything works when the program window is open. 

I received a new box last week it is different.  bluetooth device is hidden and not visible, the phone sends a hex code to it
[image]

https://a.aliexpress.com/_dY5fSFY

but they all work when the iPhone is unlocked and the application is running

tonysprenk, 2020-08-05:
Do you have an after market alarm system? What make and model car do you have?

xrust83, 2020-08-05:
I have an old car.  I installed the whole device myself.  one-way signaling system code.  the start-store button is also from aliexpress.  these devices are connected to the central lock with brown and white wires.  chain the central lock to the scales or white wire to open or close the car.  and essentially copy the button.  everything is very simple but using the phone.

Lada 2107 (1985) :-)

tonysprenk, 2020-08-05:
Ah I see. This wouldn't work
On most modern cars due to the security measures. For your car a simple esp relay should work and another esp (or raspberry pi or whatever) to create a WiFi network. Then on your phone you could make a Siri shortcut: when connected to the cars WiFi unlock the car, when disconnected lock the car.

tonysprenk, 2020-08-05:
For my car (2018 Nissan Leaf) I've been thinking about a device connected to the obd2 port. But I'm pretty sure it wouldn't start without the key inside anyway so, no point.

xrust83, 2020-08-05:
Can I do wifi from esp?  can you tell me how to do it right?

xrust83, 2020-08-05:
in this case, you need not the key itself, but just an imbilizer.  everything needs to be installed in the standard lock and the window is closed

[image]

in this case, you need not the key itself, but just an imbilizer.  everything needs to be installed in the standard lock and the window is closed

tonysprenk, 2020-08-05:
There are many guides, here's one: https://www.engineersgarage.com/tutorials/esp8266-wifi-hotspot/.

xrust83, 2020-08-05:
There are many guides, here's one:

bj-tracer, 2020-08-11:
Hi, bro!
You need to use bluetooth devices for direct connect to Home App. Wi-Fi based devices need to use a router. Even you use only one device.
And you need to use a Home hub like iPad/Apple TV/HomePod for using automations.

closed 2020-09-10 by stale[bot]