RavenSystem/esp-homekit-devices #1070: Instability since 2.5.3

macjl, 2020-08-07:
Hello,

Since I have installed the 2.5.3  HAA firmware, many of my devices hangs frequently. Sometime once or twice a day.
When a device hangs, it's not available anymore in Home App, does not respond to ping, and does not react on GPIO inputs. 
I didn't collect logs from now, as I can't predict when a device will hang.

The problem did not appear in previous firmware.

I hoped the last version which was published yesterday would help, but it doesn't. 

Do you know if there's a known issue in last firmwares?

Regards

RavenSystem, 2020-08-07:
There is a bug in last version of UDP Adv Logger. Are you using network logs.

macjl, 2020-08-07:
Yes I am on all my devices.

RavenSystem, 2020-08-07:
That is the issue. I will release a fix soon.

macjl, 2020-08-07:
Ok thanks! I'll wait for the update!

RavenSystem, 2020-08-07:
Try v2.5.5.

macjl, 2020-08-07:
Just installed on my 17 HAA devices. I will see soon if there are other issues. 

Thanks!!

macjl, 2020-08-08:
No more hangs today. The devices seems to be stable now. I close the issue. 

Thanks again for your work!

closed 2020-08-08 by macjl