RavenSystem/esp-homekit-devices #1071: Sonoff Mini not accessible via web

avatar25pl, 2020-08-08:
I installed HAA using OTA to sonoff mini. 

After that all worked and added device to HomeKit. 
However I want to change Json config but now cannot connect to Setup page anymore. 
I was flicking power without any success to restore setup screen. 
I added another one and when added to homekit it is no longer available via web? 

Is there any way to restore web access to device without flashing?

peros550, 2020-08-08:
The web access is not always on. You have to bring your device into this state. 
You can try switching your device on / off very quick for around 8 times (I always try more, it never hurts)  and it will eventually go into set up mode. In this mode, the web access is activated and your device is not responding in home app.

avatar25pl, 2020-08-08:
I tried to switch 20 times power without any luck. It is still connected to wifi but not setup page :( 
What is the point of having reset homekit ID option if I cannot even access it?

avatar25pl, 2020-08-15:
Thanks. It works finally.

closed 2020-08-15 by avatar25pl