RavenSystem/esp-homekit-devices #1072: haa v2.5.6

sercitecotf, 2020-08-08:
After updating to version 2.5.6 it does not recognize the configuration, it shows me the configuration panel and from there it does not come out, no matter how much I save
this is the json string:
  {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"r":[{"g":4,"v":1,"i":0.5}]},"2":{"r":[{"g":4,"v":1,"i":1}]},"3":{"a":2},"t":40,"d":26,"f3":[{"g":10,"t":0}],"f4":[{"g":10,"i":0,"t":1}],"b":[],"s":1,"f0":[]},{"1":{"s":[{"a":1}]},"b":[],"s":0,"i":1},{"0":{"r":[{"g":15,"v":1,"i":0.5}]},"1":{"r":[{"g":15,"v":1,"i":0.5}]},"2":{"r":[{"g":15,"v":1,"i":1}]},"t":40,"s":1,"d":12,"b":[],"f0":[],"f3":[{"g":14,"t":0}],"f4":[{"g":14,"t":1}]},{"0":{"r":[{"g":5,"v":1,"i":3}]},"t":4,"b":[],"s":1,"i":4,"f0":[]},{"0":{"r":[]},"t":6,"s":0,"i":9,"f1":[{"g":9,"t":1}]}]}

RavenSystem, 2020-08-08:
Fixed with v2.5.7.

sercitecotf, 2020-08-08:
ok
Thank you

closed 2020-09-07 by stale[bot]