RavenSystem/esp-homekit-devices #1074: unable to do OTA-Update, loops

archiolochos, 2020-08-11:
i have two blitzwolf-led-controllers, that work as wanted. One is on 2.5 and the other is on 2.5.1. I wanted to do an Update to the latest version, but it somehow did not seem to work.
so i took a look at the OTA-Logs with "nc -kulnw0 45678" and it seems not to start an update. i get the following:

`HAA Installer Version: 3.0.8

- Server: github.com/RavenSystem/haa/releases/latest/download
- Port:   443
- SSL:    yes

Restarting...
192.168.2.36-A3608C-: 192.168.2.36-A3608C-: 
192.168.2.36-A3608C: HAA Installer Version: 3.2.0
192.168.2.36-A3608C: 
192.168.2.36-A3608C: - Server: github.com/RavenSystem/haa/releases/latest/download
192.168.2.36-A3608C: - Port:   443
192.168.2.36-A3608C: - SSL:    yes
192.168.2.36-A3608C: 
192.168.2.36-A3608C: 
192.168.2.36-A3608C: Restarting...
192.168.2.36-A3608C: 
192.168.2.36-A3608C: HAA Installer Version: 3.2.0
192.168.2.36-A3608C: 
192.168.2.36-A3608C: - Server: github.com/RavenSystem/haa/releases/latest/download
192.168.2.36-A3608C: - Port:   443
192.168.2.36-A3608C: - SSL:    yes
192.168.2.36-A3608C: 
192.168.2.36-A3608C: 
192.168.2.36-A3608C: Restarting...
192.168.2.36-A3608C: 192.168.2.150-C9AB5A-: 192.168.2.150-C9AB5A-: 
`

Is there any Idea what goes wrong and how to perform an update?

spam667, 2020-08-11:
Same here... I am trying to update a Shelly1 and I am into a loop. If I capture traffic with "nc -kulnw0 45678" I can see a loop. The device picks the file "otamain.bin" but then it says "Restarting..." and the process starts again.

I attach the log in case it is useful.

[otamain.txt](https://github.com/RavenSystem/esp-homekit-devices/files/5058359/otamain.txt)

spam667, 2020-08-12:
Hi again! I solved the problem :) try this archiolochos, at least it worked for me.

It seems that there are two different things into the HAA: Installer and Main.

In my case, I "flashed" the Shelly 1 vía OTA using a link from the internet, in my case the link was this one:

http://X.X.X.X/ota?url=http://dl.dasker.eu/firmware/mg2haa-Shelly1.zip

Problem with this procedure was that the mg2haa uses a local copy of the HAA Installer and it is still using the 3.2.0 version. The HAA Installer 3.2.0 version is only compatible with HAA Main versions until 2.5.2. Since 2.5.3 this Installer is no longer compatible and the installation will fail and enter into a loop.

Solution for me was the following:

1º) I set my Shelly1 into setup mode.
2º) I accessed the menu and under the "Custom Server for updates" I have set this link:
github.com/RavenSystem/haa/releases/download/2.5.0/
3º) Wait a bit and you will have version 2.5.0 installed of HAA Main, with HAA Installer 3.2.0.
4º) Access again to "Setup Mode" and select "Update all now (HAA OTA v3.2.0)"
5º) Wait a bit and you will end with your device updated to HAA Installer 4.0.0 and HAA Main 2.5.7.

This worked for me! I hope it helps to anyone who has the same problem.

BR!

spam667, 2020-08-12:
> i have two blitzwolf-led-controllers, that work as wanted. One is on 2.5 and the other is on 2.5.1. I wanted to do an Update to the latest version, but it somehow did not seem to work.
> so i took a [...]

Probably you have to check if your installer 3.0.8 is compatible with the version you are trying to install... If not, you can download and old version and build your own HTTP server to point your device to this server and download the old version.

archiolochos, 2020-08-13:
Hello spam667,

i tried your method but it did not work for me. i took a look at the link you provided (github.com/RavenSystem/haa/releases/download/2.5.0/) and tried to open it in my browser - i get an 404 error. it should deliver something i guess?

when i want to do my own server i would need the binaries - can you tell me where i can get those?

thanks so far :-)

closed 2020-08-13 by archiolochos

spam667, 2020-08-13:
Hello archiolochos,

there is some king of correlation between the HAA Installer and the HAA Main, I tried to explain on my last post, sorry if I was not completely clear.

For example HAA Installer 3.2.0 works with HAA Main versions until 2.5.2. HAA Installer 4.0.0 works with versions from 2.5.3 to 2.5.7.

In your case you have HAA Installer 3.0.8, I don't know which one will be the right version for that installer. Maybe the developer could help in here. Also I tried to access to the version history and the last one I saw was 2.4.7.

Don't worry if the link shows you a 404 error. The Installer is looking for a file, for example otamain.bin, so if you want to check the link in the browser you should access to something like this https://github.com/RavenSystem/haa/releases/download/2.5.0/otamain.bin and you will see that the file starts downloading to your PC/Mac. But for the HAA Installer to work you should configure the link with the correct format, like this: github.com/RavenSystem/haa/releases/download/2.5.0/

try changing 2.5.0 by 2.4.7 and see if this works, If not, maybe you'll have to ask the developer for an older version or flash your device again using a FTDI and a serial cable.

Good luck! :)

spam667, 2020-08-13:
I have tried and you can also use this link:

https://github.com/RavenSystem/haa/releases/download/2.4.0/otamain.bin

So there are older versions than 2.4.7.

spam667, 2020-08-13:
Ok, sorry again.

If you go here you can check all the version history:

https://github.com/RavenSystem/esp-homekit-devices/releases?after=HAA_2.4.2

The HAA Installer 3.0.8 was developed last 21-may and version 2.3.7 next 25-may. Maybe you should try with version 2.3.7 to 2.3.10 that were the ones released between the Installer 3.0.8 and the installer 3.1.0.

I don't have clear if you have 3.0.8 or 3.2.0. Because in your log you can see at the beginning HAA Installer 3.0.8 but later in the logs it seems to be HAA Installer 3.2.0.

You can change the link accordingly. You should be able to install those versions with your current OTA Installer but check the bugs/caveats opened for these old version not to brick your device :) I hope when you install this version, you can check later the option to "update all" and get to the latest one with no problem.

I hope this helps!