RavenSystem/esp-homekit-devices #1079: shelly 2.5 as a roller shutter with external pushbuttons needs a double click

Leonoerlemans, 2020-08-14:
Hi,

Still most grateful for this firmware.
How come that I need to push my momentary switch twice to get it into action (as well up as down). When I update by pushing the buttons 8 times and then wait, that problem is gone, but after a day or so the problem returns and I need to do it again.
Have you a solution?

I used  the exact json from the provided device database

Kind regards,

Leon

Leonoerlemans, 2020-09-10:
Hi!

Does anyone else have or had  this problem or has a solution?

Kind regards

Leon

Leonoerlemans, 2020-10-06:
@RavenSystem I am sorry to nag about it, but do you have an idea what is causing this problem?

Kind regards,

Leon

RavenSystem, 2020-10-12:
I am not able to replicate your issue.

closed 2020-11-12 by stale[bot]