RavenSystem/esp-homekit-devices #1081: Problem to load info from HAA device using iOS app Controller for HomeKit

pilot1981, 2020-08-15:
Hi,
I’m using iOS controller for HomeKit to backup HomeKit configuration:

https://apps.apple.com/it/app/controller-for-homekit/id1198176727

But this app isn’t able to load information related to HAA devices...

There is some standard/protocol not already deployed In HAA firmware?

peros550, 2020-08-15:
in case you haven't tried, it would be a good idea to also contact the developer of the app about this. I remember I had a question about the way the backup function used to work and emailed them. They responded within a few days.

pilot1981, 2020-08-16:
Thanks @peros550, I already contacted app developer, I’m waiting for an answer...

pilot1981, 2020-08-28:
Hi @RavenSystem,
Developer answered:

“I guess the problem with HAA is that there is no serial number set.”

It’s possible have serial number set in our HAA devices?

RavenSystem, 2020-08-28:
HAA devices have serial number. You can see it under accessory details.

pilot1981, 2020-09-06:
I’m waiting for update from app developer

pilot1981, 2020-09-23:
App developer don’t answer...

I notice app read serial number of dummy switch HAA and official HomeKit devices, but not for real HAA devices...

I don’t understand...

pilot1981, 2020-10-02:
@RavenSystem you think last version 3.5 can fix this problem? Changing the use of serial number?! I will try...

pilot1981, 2020-10-03:
Issue related to serial number problem. Solved in HAA version 3.5

closed 2020-10-03 by pilot1981