RavenSystem/esp-homekit-devices #1082: No wifi hotspot. Sonoff basin R3

toddc01b, 2020-08-16:
I’m new to this.
 I have just tried to flash Sonoff basic R3 using NodeMCU PyFlasher and the latest “fullhaaboot.bin”. It says “firmware successfully flashed” but once I re-apply power I can’t seem to find the wifi hotspot for the Sonoff. Also the wifi light on the Sonoff remains off. 
Any help would be much appreciated.

Murdrock, 2020-08-20:
The light remains off that's normal. You don't see the HAASOMTHING hotspot;

toddc01b, 2020-08-20:
Thanks for the reply. 
Yes that's correct. I can't see the hotspot which means I can't configure it and connect it to my wifi. 
Any suggestions?

Murdrock, 2020-08-20:
Well it seems like a dead end... maybe someone with more experience could help. You seem to do the steps correct....

Murdrock, 2020-08-20:
Any screen shoot of the flash program before flahsing?

closed 2020-09-20 by stale[bot]