RavenSystem/esp-homekit-devices #1087: Sonoff SV - Old Doorbell 16VAC

lfiocco, 2020-08-18:
Hi there. I would like to convert my old "dumb" 16vac door bell into a smart one.

Here is the plan.
Sonoff SV
Old doorbell button connected to GPIO14
Old transformer and ding-dong connected to power and open circuit to sonoff relay.
Configured as a sensor in HAA.

When the doorbell is pressed, GPIO14 is closed, it activates the relay for 2 seconds, the ding-dong is heard.
Homekit sees the alert and triggers automation.

doable? what accessory would best fit this plan?

shanghaix, 2020-08-20:
Yes sonoff SV works, that’s how my old doorbell is set up. You can also use ffmpeg plugin in homebridge to get a notification with a picture of whoever rang if you have a camera.

lfiocco, 2020-08-20:
my door bell runs on 16VAC. From what I read, I just remove the jumper and feed the AC power on the inputs. When triggered it will output the same voltage.

Is this your setup?

lfiocco, 2020-08-24:
what does your config look like?

closed 2020-09-23 by stale[bot]