RavenSystem/esp-homekit-devices #1089: Tuya GU10 bulbs

Ie0nard0, 2020-08-19:
Hi! I’d like to flash some gu10 tuya bulbs, can someone help me with json? This is the blakadder info:
title: Generic GU10 5W 460lm
link: https://www.ebay.de/itm/5W-GU10-LED-Smart-WIFI-Birne-Glühbirne-RGBW-Dimmbar-für-Alexa-Google-Home-DE-DHL/124139389928
image: /assets/images/generic-gu10-5w-rgbwcct.jpg
template: '{"NAME":"RGBCCT GU10","GPIO":[0,0,0,0,37,40,0,0,38,41,39,0,0],"FLAG":0,"BASE":18}'
flash: tuya-convert
category: bulb
type: RGBCCT
standard: gu10

Thanks

bj-tracer, 2020-08-20:
Aloha!
What kind of controller applied at this bulb?
Do you have internal fotos?

Ie0nard0, 2020-08-20:
> Aloha!
> What kind of controller applied at this bulb?
> Do you have internal fotos?

I just opened up one of them. It’s an ESP8266EX as you can see in the picture. Thank you. 
[image]