RavenSystem/esp-homekit-devices #1090: magic home rgb controller JSON problem

exrector, 2020-08-19:
The standard code works but has a mix of G and V colors:
{"c": {"q": 850,"b": [{"g": 0,"t": 5}]},"a": [{"t": 30,"r": 5,”g”: 12,”v”: 13, ”b”: [{"g": 0}]}]}
I only changed two digits and the device does not appear to connect to homekit:
{"c": {"q": 850,"b": [{"g": 0,"t": 5}]},"a": [{"t": 30,"r": 5,”g”: 13,”v”: 12, ”b”: [{"g": 0}]}]}

**What is the problem????????**

gonzabur, 2020-08-19:
You have a problem with the semicolons. Try to use plain text.

xrust83, 2020-08-19:
Возможно кавычки от пустой раскладки не подходят ;-)

exrector, 2020-08-19:
Thanks to everyone who responded and gave me hints. I've removed all the spaces. Will change only numerical values. The code worked!

`{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":5,"g":12,"v":13,"b":[{"g":0}]}]}`

xrust83, 2020-08-19:
Так ото ж

closed 2020-08-19 by exrector