RavenSystem/esp-homekit-devices #1091: Sonoff basic R2 motion sensor

adamkasprzak, 2020-08-20:
Hello!

I've been reading most of the threads and wiki on how to configure a motion sensor with sonoff basic R2. Till now I can not get it to work properly. What I want to achieve is to get motion sensor turn GPIO12 (replay) on when active, then after 10 seconds turn it off. This is the exact scenario as in wiki (https://github.com/RavenSystem/esp-homekit-devices/wiki/Sensors#example)

I have my PIR (HC-SR501 or HC-SR505 - tested both) connected to RX (GPIO3 on Sonoff basic R2) and my code is:

```
{
  "c": { "l": 13},
  "a": [{
    "t": 12,
    "i": 10,
    "0": { "r": [{ "g": 12, "v": 0 }] },
    "1": { "r": [{ "g": 12, "v": 1 }] },
    "f1": [{ "g": 3 }]
  }]
}
```

I don't know what's wrong. PIR is connected to 5V from Sonoff board and data pin in connected to GPIO3 (RX). Today the package with AM312 PIR is arriving and will test it with 3.3V from Sonoff.

Thank for any help!
Adam

vsidlu, 2020-08-20:
Hi,

This is my json for motion - sonoff basic.

{"c":{"o":1,"l":13,"i":1,"n":"dresingroom_motion","b":[{"g":0,"t":5}]},"a":[{"t":2,"s":5,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":5,"f1":[{"g":14,"p":0,"i":1,"t":0}]}]}

There is t in motion accessory which I can not explain but it helps.

Try this one

adamkasprzak, 2020-08-24:
Hi! The reason for outlet accessory is to have ability to manually turn on the light if necessary. I plan to use this combo as a light outside the house.

What I find strange is that sometimes the motion sensor triggers the relay and sometimes is not. What is the algorytm in general? I have red that the sensor checks every 2s if there is a motion and gives 3.3v on data pin - I have checked that. 

Why sonoff is not executing "1" when there is 3.3v on GPIO3 (RX) after it changed to "0"? Is there another delay?

adamkasprzak, 2020-08-24:
Btw, Murdrock, thanks for the tip on how to convert 5V HC-SR501 to 3.3V HC-SR501. For others - you simply connect 3.3V as  Murdrock showed by the arrow, then Ground and Output!

Murdrock, 2020-08-24:
{"c":{"l":13,"n":"Motion","b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":16,"t":1},{"g":16,"t":0}],"s":0},{"1":{"s":[{"a":1}]},"b":[],"s":0},{"1":{"s":[{"a":2}]},"b":[],"s":0},{"t":12,"s":0,"f0":[],"f1":[{"g":4,"t":1}],"i":6}]}

This is the one I use for mine. Gpio 16 (jumper pins) is used to trigger the switch manually, and the motion triggers it through Home app automation.

The motion sensor is connected to s1s2.

Two dummy switches for setup And OTA.

adamkasprzak, 2020-08-24:
Murdrock, everything work, except motion sensor :) Have you connected data to s2 and ground to s1?

adamkasprzak, 2020-08-24:
I think I have figured out the HC-SR501 PIR. It works when I set the time delay on the sensor to minimum (0,3s?). It would be veery helpful to know the algorithm though... Thanks guys for your help!

Murdrock, 2020-08-24:
Data to s2 and ground s1.st the manual delay on PIR to minimum.

adamkasprzak, 2020-08-24:
Thanks, sitting the delay to minimum was the key!

Murdrock, 2020-08-24:
I don't know what else to think. I am not an expert! Just tinkering stuff!

closed 2020-09-23 by stale[bot]