RavenSystem/esp-homekit-devices #1099: Sonoff Default on and the switch close for 10sec

junkerp, 2020-08-23:
is it possible to invert the button on the switch?

I need the sonoff switch to reset my server.
The switch is default on, and when i hit the switsh-buttin, i like the switch to be closed in 10sec.
 My settings for now is this...
{
  "c": {"l": 13 },
  "a": [{
    "t": 1,
    "0": { "r": [{ "g": 12, "v": 1  }] },
    "1": { "r": [{ "g": 12, "v": 0 }] },
    "b": [{ "g": 0 }],
    "s": 0,
    "i": 10
  }

It is working, but i dont like that the icon in Apple homkit shows closed, when the switch is on. 

regards Peter

Mattis-Schulte, 2020-08-23:
"0": { "r": [{ "g": 12, "v": 0 }] },
"1": { "r": [{ "g": 12, "v": 1 }] }

Like this?

junkerp, 2020-08-23:
Hi Mattis

the lines you send set the switch default on, but close it after 10sec  and stays off 
i need the switch to be default on and if i pres the button close in 10sec and then turn on and stays on.

the line i use is.
"0": { "r": [{ "g": 12, "v": 1 }] },
"1": { "r": [{ "g": 12, "v": 0 }] },
it is working fine, but the icon in Apple homkit shows off when it is on. :-)

closed 2020-09-22 by stale[bot]