RavenSystem/esp-homekit-devices #1100: REQUEST: Sonoff SV as a simple "push button" for opening gate

Gejza75, 2020-08-24:
Can someone generate JSON for **Sonoff SV** which work as a simple push button - ON/OFF in a 0,5 seconds, please? No additional sensors, nothing. Just simple push button. Thank you very much.

NikDevx, 2020-08-28:
> Can someone generate JSON for **Sonoff SV** which work as a simple push button - ON/OFF in a 0,5 seconds, please? No additional sensors, nothing. Just simple push button. Thank you very much.

https://github.com/RavenSystem/esp-homekit-devices/wiki/Stateless-Button

You can user configurator https://glumb.github.io/haa-configurator/

Gejza75, 2020-08-30:
Thanks Nik. Tried but without success.. I tried to modify Garage Door, but it works just partly. I would be greatful, if someone would help me with that.

Thank you very much for any help!

QDUNI, 2020-08-31:
Do you mean a switch ?

QDUNI, 2020-08-31:
Do you mean a physical push button? Or a push button in the home app?

QDUNI, 2020-08-31:
Try this config of a stateless button: 

`{
  "c": {
    "b": [{ "g": 0, "t": 5 }]
  },
  "a": [{
    "t": 3,
    "0": {"r":[{"g":12,"v":1,"i":0.5}]},
    "f0": [{ "g": 0, "t": 1 }]
  }]
}`

**In one line:**

`{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":3,"0":{"r":[{"g":12,"v":1,"i":0.5}]},"f0":[{"g":0,"t":1}]}]}`

Gejza75, 2020-09-09:
Hi all. After couple of testing and playing I found formula. And finally it works like I wanted. Here it is:

{"c": {"l": 13,"b": [{"g": 0,"t": 5}]},"a": [{"0": {"r": [{"g": 12}]},"1": {"r": [{"g": 12,"v": 1}]}, "b": [{"g": 0,"t": 2,"p": 0,"i": 0}],"s": 0,"i": 0.2}]}

Thanks to all, who try to help. If there is possibility, please put it in device database.

Thank you.

closed 2020-10-10 by stale[bot]