RavenSystem/esp-homekit-devices #1103: How many accesories into a esp8266?

edupulu, 2020-08-26:
Hello! I cannot able to find if this question is already answered.
The question is, what is the limit of accesories can be configured (with json) in a esp8266 (esp07s)?
I have developed a board with 12 channels of drivers to manage water valves and it would be amazing if I could use your system.
Thanks!!

edupulu, 2020-09-06:
Any suggestion?

tonysprenk, 2020-09-06:
In theory 150 because that's the limit HomeKit sets. But HAA is also limited by the memory of the esp chip. If you want to be sure that it's going to work use an esp chip with higher amount of memory available. I would probably just try with 1 esp module and if that overloads add another.

edupulu, 2020-09-09:
Ok thanks. I'll try what is the limit in my case.

closed 2020-10-10 by stale[bot]