RavenSystem/esp-homekit-devices #1104: Garage door with NC reed contacts

morphix9, 2020-08-27:
Hey guys,

Sonoff SV with 2 magnetic reeds - both NC (normally closed).

This is an example, where it was mentioned that it's for 2 NO sensors:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},
{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"t":40,"d":31,"f3":[{"g":14,"t":0}],"f4":
[{"g":14}],"f2":[{"g":4,"t":0}],"f5":[{"g":4}]}]}

Do I need to change anything to work with NC instead of NO?

Thanks for your help!

rjblake, 2020-08-28:
Think you need to change the "t":0 to "t":1 (under "f3" and "f2")

morphix9, 2020-08-29:
> Think you need to change the "t":0 to "t":1 (under "f3" and "f2")

And what about f4 and f5? These should then be set to “t”:0?

closed 2020-09-28 by stale[bot]

morphix9, 2020-09-28:
Still having issue with this garage door accessory using magnetic reeds. Can't figure it out. Sometimes it's working, sometime not. I have 30+ other accessories working fine, but the garage door is strange.

@RavenSystem can you confirm I can use NC sensors instead of NO and how the "t" has to be set for such a case?