RavenSystem/esp-homekit-devices #1108: Sonoff Basic TH16: does not start emergency mode

coyotanark, 2020-08-30:
Hi!

Long story, short this will explain the other issue I open too:
- unexpected router change, setup new wireless network with new SSID.
- try to enable emergency mode on multiple Sonoff TH16: power on, quickly power off (within 3 seconds) then power on again

What is expected:
- wifi to connect to as the sonoff should be in Access Point Mode
- update wifi settings and the device will find back normal working state. 

What happens:
- unsure but no wifi network to connect to. This affects 3 different devices, setup with 3 slightly different configurations and 3 different firmware versions (I see in the home app from when they were still there)

Kitchen ventilation:
https://ibb.co/4YVtZtk

Firmware 2.4.6
config:
{
  "c": { "l": 13 },
  "a": [{
    "t": 1,
    "0": { "r": [{ "g": 12, "v": 0  }] },
    "1": { "r": [{ "g": 12, "v": 1 }] },
    "b": [{ "g": 0 }],
    "s": 0
  }, {
    "t": 24,
    "g": 14,
    "n": 4,
    "j": 0.5,
    "y1": [{
      "v": 80,
      "r": 1,
      "0": { "r": [{ "g": 12, "v": 1 }] }
    }, {
      "v": 79,
      "0": { "r": [{ "g": 12, "v": 0 }] }
    }]
  }]
}

Bathroom ventilation (setup like 3 weeks ago)
https://ibb.co/C1wbChb

Firmware 2.5.7
config:
{
  "c": { "l": 13 },
  "a": [{
    "t": 1,
    "0": { "r": [{ "g": 12, "v": 0  }] },
    "1": { "r": [{ "g": 12, "v": 1 }] },
    "b": [{ "g": 0 }],
    "s": 0
  }, {
    "t": 24,
    "g": 14,
    "n": 4,
    "j": 0.5
  }]
}

£rd one I really needed it so I ended opening it and flashing it. it was under 1.x something version. It is now working fine on the new network, proves the network is ok.
Config:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":22,"g":14,"n":3},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

How can I change the wifi of the 2 others without getting into my ceiling with the laptop please?

Thanks

coyotanark, 2020-08-30:
FYI, here is the router connected device showing the only one connected, the one I flashed and no more:

https://ibb.co/c8s8PVX

coyotanark, 2020-09-25:
Hey I don't want this issue closed but solved. Too easy to ignore it until the bot closes it. There are many other instances of similar reports in the issue list that were ignored and closed by the bot.

RavenSystem, 2020-10-12:
Emergency setup mode works fine with all versions.

closed 2020-10-12 by RavenSystem