RavenSystem/esp-homekit-devices #1109: Sonoff basic R1: does not connect to the new wifi when settings updated.

coyotanark, 2020-08-30:
This is a different issue than the other one I posted but arise in the same context:
https://github.com/RavenSystem/esp-homekit-devices/issues/1108

So as detailed in the other issue, new network. I have multiple Sonoff Basic R1 setup all around the house and I can't change the wifi on any of them.

Long story, short this will explain the other issue I open too:
- unexpected router change, setup new wireless network with new SSID.
- try to change wifi settings in setup mode for many son off basic, they never connect to the new wifi.

They all share the same or almost same config (some are just on and some other off by default when powered on):
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":14},{"g":14,"t":0}]}]}

Various firmwares as they were installed over multiple month (1+ year), which BTW is strange as auto OTA update should have updated them, but no... another issue.
1.8.6 x7
2.5.2 x1
2.5.1 x1

Same than the other post, how do I get them to work without removing every appliance / getting into my ceiling?

Thanks

coyotanark, 2020-09-25:
Hey I don't want this issue closed but solved. Too easy to ignore it until the bot closes it. There are many other instances of similar reports in the issue list that were ignored and closed by the bot.

ITuranchoks, 2020-10-12:
I have the same problem.

coyotanark, 2020-10-12:
There are various other similar issues already reported that were never answered and closed marked as stalled. I guess it is useless to spend time reporting issues here, the lack of acknowledgment feels like it is abandoned. I don’t expect anything anymore and I’d advise to move to some more reliable solution. However I’ll keep this issue opened as it is an ongoing not addressed issue and it is wrong to close it.

coyotanark, 2020-11-07:
This is still an issue and not resolved

RavenSystem, 2020-11-09:
I am not able to reproduce your issue. When I change wifi settings, device connects to new one.

coyotanark, 2020-11-09:
Good news, I can reproduce it. How can I collect logging from the device considering I cannot connect it to a new wifi?

RavenSystem, 2020-11-09:
Use UART to collect logs.

coyotanark, 2020-11-09:
Can you please refer to a resource on how to do it? I’d rather get ready before getting my laptop plugged to something in my ceiling. Note sure if it would vary across versions, but it would probably run 1.8.6

RavenSystem, 2020-11-09:
First update to last version.

coyotanark, 2020-11-09:
And how do I do it without connecting it? Not to mention the update that broke the git update. So basically you mean to flash the device and erase any trace of the issue, right? There is no issue then as you erase everything and keep pretending all works fine...

RavenSystem, 2020-11-09:
First: There is not any trace of any issue.
Second: To find a bug in an older version does not make sense if last version has it fixed.

closed 2020-11-09 by RavenSystem