RavenSystem/esp-homekit-devices #1113: temperature and humidity association for relay activation

clint80, 2020-09-03:
Hello . I would like to control a CMV with a DTH11 sensor. The temperature and humidity should be taken into account. I can explain to myself if a temperature set at 27 ° and humidity set at 85%. Both conditions must be met to activate the relay.
if T has "27 °" H has "80%" relay no. if T "25 °" and H "85%" relay no. if T has "27 ° or +" and H has "85% or +" relay ok.
I did not find the function to be able to program. here is my json which works with the first function reached. {"c": {"l": 13}, "a": [{"t": 24, "n": 1, "g": 0, "j": 10, "h": - 0.5, "y0": [{"v": 27, "r": 1, "0": {"r": [{"g": 1, "v": 1}]}}, {"v": 25, "0": {"r": {"g": 1, "v": 0}]}}], "y1": [{"v": 85, "r": 1, "0" : {"r": [{"g": 1, "v": 1}]}}, {"v": 75, "0": {"r": [{"g": 1, "v ": 0}]}}]}, {" i ": 0.5," 1 ": {" s ": [{" a ": 1}]}}]}.
Is there a function that allows you to do this if yes could you help me achieve it?
a big thank you for all the work provided which allows us to improve our ordinary

sorriso93, 2020-09-21:
Hello you can use EVE homekit to do this with events

clint80, 2020-09-25:
hello
I tried but I do not have access to the temperature and humidity icon
is there a hub (ipad, tv4 or other) to be able to create events
Thank you for your reply

RavenSystem, 2020-09-25:
Don't use external automations to do it. Better use wildcard actions. Example:

clint80, 2020-09-25:
Hello
I followed suit. but the relay is activated as soon as one of the actions is reached. either temperature or humidity.
I would like the temperature and humidity to be taken into account at the same time. ie 27 ° and 85%, the relay trips and if either drops, the relay returns to its initial position. if either reaches its trigger level, the relay does not change position.
I haven't found a solution yet
thank you for taking some of your time to answer me.
and a big thank you again for all the work done.

sorriso93, 2020-09-28:
> hello
> I tried but I do not have access to the temperature and humidity icon
> is there a hub (ipad, tv4 or other) to be able to create events
> Thank you for your reply

In EVE you define a new rule and select the sensor you want as event. The hub I think is needed to use homekit from remote
[image]

clint80, 2020-09-29:
Hello
in EVE, if I go to SCENARIO, the temperature sensors do not appear.
if I go to RULES, it tells me that I need a concentrator

clint80, 2020-10-25:
Hello
it would be possible to add a function in the json program that takes into account the temperature and humidity to trigger a relay
thank you for all the work done

clint80, 2020-11-23:
Hello
RavenSystem according to what you wrote, I would have to add to my json the commands "p0" and "p1" to take into account the 2 values at the same time to trigger the relay
https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#wildcard-action-example-1

closed 2020-12-24 by stale[bot]