RavenSystem/esp-homekit-devices #1114: After flash on ESP32, no HAA-XXXX wifi accesspoint

Castle11, 2020-09-03:
Hi there, 
I'm currently studying how to code and this is my first home automation project.
I have followed the three steps of the wiki after erasing flash and flashing new firmware, nothing happened...
What is really strange to me is the fact that the board is actually able to emit an AP signal with a classical Arduino 'WiFiAcessPoint' example (given by the Arduino soft itself !!) an I'm able to communicate and blink a board's LED wirelessly...

I am completely lost, If somebody could help me with this issue I'l be glad !

here are my current hardware/software details and tests : ESP32-D0WDQ6 (revision 1), Python 3.8 (PyCharm) on Mac OS (High Sierra)
after erasing : 'Chip erase completed successfully in 8.8s'
after flashing fullhaaboot.bin : 'Wrote 401388 bytes (284499 compressed) at 0x00000000 in 25.2 seconds (effective 127.5 kbit/s)...
Hash of data verified.'

Am I missing something ? 
What I've tried so far : 
-doing the same process with python 2.7
-powering my esp32 with +3.3 & 5V (via arduino uno)
-flashing the esp32 card by pressing booth 'EN' and 'BOOT' buttons (then releasing BOOT, then releasing EN) 

And as expected by the Murphy's law, Nothing happened ...

wrx5, 2020-09-03:
I have the same problem with ESP8285 chip and another issue for someone else is still open for the same reason. I don't think we are doing anything wrong. 

Download the tasmota.bin and flash that one, you will see it will work. 

Last 3-4 days nothing left that I haven't tried. Whatever solution I saw on the Internet I tried but no luck. I have One last thing left for me to try is previous version of the fullhaaboot.bin Latest version might have an issue

wrx5, 2020-09-03:
ohh by the way, I am not sure if ESP32 uses flash mode DOUT but if you haven't tried yet try it. -fm dout or --flash_mode dout. 

If this doesn't work I would try other flash modes.  QIO, DIO and QOUT

put it into flash/setup mode again
esptool.py -p <COM_port> erase_flash

reboot your device and put it into flash/setup mode again
esptool.py -p <COM_port> --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 fullhaaboot.bin

tonysprenk, 2020-09-03:
HAA doesn't support esp32 AFAIK.

rjblake, 2020-09-04:
> Hi there,
> I'm currently learning how to code and this is my first home automation project.
> I have followed the three steps of the wiki after erasing flash and flashing new firmware, nothing happened...
> What is really strange to me is the fact that the board is actually able to emit an AP signal with a classical Arduino 'WiFiAcessPoint' example (given by the Arduino soft itself !!) an I'm able to communicate and blink a board's LED wirelessly...
> 
> I am completely lost, If somebody could help me with this issue I'll be glad !
> 
> here are my current hardware/software details and tests : ESP32-D0WDQ6 (revision 1), Python 3.8 (PyCharm) on Mac OS (High Sierra)
> after erasing : 'Chip erase completed successfully in 8.8s'
> after flashing fullhaaboot.bin : 'Wrote 401388 bytes (284499 compressed) at 0x00000000 in 25.2 seconds (effective 127.5 kbit/s)...
> Hash of data verified.'
> 
> Am I missing something ?
> What I've tried so far :
> -doing the same process with python 2.7
> -powering my esp32 with +3.3 & 5V (via arduino uno)
> -flashing the esp32 card by pressing both 'EN' and 'BOOT' buttons (then releasing BOOT, then releasing EN)
> 
> And as expected by the Murphy's law, Nothing happened ...

Since you have a Mac, try using **NodeMCU PyFlasher** (has GUI and simple to try/change options). As @wrx5 says, think the ESP32 uses DIO, not DOUT (used for ESP8285). However, as @tonysprenk mentions, ESP32 is not supported by HAA as seen here in [this thread](https://github.com/RavenSystem/esp-homekit-devices/issues/877)

Castle11, 2020-09-04:
> I have the same problem with ESP8285 chip and another issue for someone else is still open for the same reason. I don't think we are doing anything wrong.
>
> Download the tasmota.bin and flash that [...]

 @wrx5 thanks for the tip ! I've tried this morning by flashing tasmota.bin but unfortunately that also does not work with my ESP32 chip. After a quick overvew of their github page ( https://tasmota.github.io/docs/Getting-Started/ ) I presume tasmota is only available for ESP8285 chips (and derivates).
By the way, if HAA is compatible nowadays with ESP32, fullhaaboot.bin latest version might effectively have an issue ? Maybe @RavenSystem have some clue about that?

> ohh by the way, I am not sure if ESP32 uses flash mode DOUT but if you haven't tried yet try it. -fm dout or --flash_mode dout.
>
> If this doesn't work I would try other flash modes. QIO, DIO and [...]

Tried also with QIO, DIO and QOUT but no SSID so far :/

Castle11, 2020-09-04:
> HAA doesn't support esp32 AFAIK.

> Since you have a Mac, try using **NodeMCU PyFlasher** (has GUI and simple to try/change options). As @wrx5 says, think the ESP32 uses DIO, not DOUT (used for ESP8285). However, as @tonysprenk [...]

@tonysprenk and @rjblake, thank you for your answers. Effectively, I just went trough this thread and saw the last comment from @RavenSystem before closing this issue. I think the only solution for me is to invest in an ESP8285 or wait for an ESP32compatible-HAA update ?

Anyway thanks also @rjblake for the NodeMCU PyFlasher advice (I'm downloading it).

closed 2020-10-04 by stale[bot]