RavenSystem/esp-homekit-devices #1116: iOS 14 beta - remote access issue

angyalZK, 2020-09-05:
Just a perception: Remote access stopped working; from local network was ok, but from cellular failed, Home app said no home center available. Automations also stopped working. (iPhone: iOS 14 beta, TV: tvOS 14 beta; TV showed as Home Center itself)
The problem NOT with HAA, the new iCloud a little bit buggy. 
Solution: turn off iCloud on TV and iPhone also and turn back. 
All devices and automations working well again. 
PS: iOS 13 showed all fine, but automations not worked.

bj-tracer, 2020-09-10:
Hi!
I use beta at iphone and atv too. But it's working properly.
Have a bug only with two Rubetek power sockets - Home app can't connect to this sockets but Rubetek app work fine.
https://rubetek.com/catalog/osveshenie_i_elektropitanie/umnaya-wi-fi-rozetka/
This devices can work with HomeKit from the box.

All HAA-devices haven't problems at local and cellular network.

coyotanark, 2020-09-11:
I have been running all betas on HomePod, Apple TV and iOS devices, no issues beside Siri loosing track of room names so far

closed 2020-10-11 by stale[bot]