RavenSystem/esp-homekit-devices #1117: Using Sonoff TH10 as a thermostat with humidity sensor

tlagarto, 2020-09-05:
Hi, I'm having trouble in configuring a Sonoff TH10 as a thermostat with humidity sensor (Si7021) and with two extra accessories for quick access to config and OTA update.
Here's the json string I'm using, what am I doing wrong?

`{"c":{"o":0,"l":13},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":0}]},"3":{"r":[{"g":12,"v":1}]},"t":25,"b":[],"g":14,”n”:4,”w”:3,”j”:60,"s":0,"z":0,"h":0,"m":15,"x":25,"d":3,"f0":[],"f1":[],"f3":[]},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0},{"1":{"s":[{"a":2}]},"t":2,"i":0.5,"b":[],"s":0}]}`

thanks

vsidlu, 2020-09-05:
Chech brackets - 14,”n”:4,”w”:3,”j”:60 - must be all equal!

tlagarto, 2020-09-16:
Thanks, that's a typo I did when copying here. But the problem persists even with the correct quotes character: ".
The issue it seems to be that the system does not recognise the device type ("t":25). Any help?

tlagarto, 2020-09-16:
After some more digging I got this one to work:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":0},{"g":12,"v":1}]},"5":{"a":0},"t":25,"s":5,"w":3,"n":4,"j":60,"g":14},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0},{"1":{"s":[{"a":2}]},"t":2,"i":0.5,"b":[],"s":0}]}`
It's a thermostat (for heating and cooling) on a Sonoff TH10 with sensor Si7021 and two extra accessories for quick access to config and quick update.

closed 2020-10-16 by stale[bot]