RavenSystem/esp-homekit-devices #1120: Semi smart switch unresponsive since 3.1.1

L4RGER, 2020-09-07:
Hey, all of my zemismart switches(2 and 3gang) updated as part of an automation on Sunday night (last night). This morning the top button on them is unresponsive until the bottom button is pushed and then it works perfectly. They have been installed for quite some time without issue. 
I will post the 3 gang config below. And a link to a video demo of what is happening. I’ve enabled logs and there is no output what so ever when the switch is in responsive. Only after one of the lower switches is toggled. After a few minutes it goes back to being unresponsive 
[Unresponsive demo](https://imgur.com/a/iHamWmL) 
`{"c":{"n": "kitchen 3gang","l":2,"i":0,"b":[{"g":4,"t":5}]},"a":[{"0":{"r":[{"g":13,"v":0}]},"1":{"r":[{"g":13,"v":1}]},"t":1,"b":[{"g":16,"t":1}],"s":5},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[{"g":5,"t":1}],"s":5},{"0":{"r":[{"g":14,"v":0}]},"1":{"r":[{"g":14,"v":1}]},"t":1,"b":[{"g":4,"t":1}],"s":5},{"i":0.5,"1":{"s":[{"a":1}]}}]}`

RavenSystem, 2020-09-07:
Fixed with v3.1.2.

wrx5, 2020-09-08:
I have the same switch. I don't get HAA wifi after flashing. How did you do yours?

L4RGER, 2020-09-08:
Issue resolved with latest release. I flashed with serial cables. 
The manufactures put some really nice labelled sockets on the pcb for us. How are you trying to flash, and does it say it’s successful?

closed 2020-09-08 by L4RGER

wrx5, 2020-09-08:
I am flashing with serial cables. FTDI thing. if I flash tasmota. bin it works. Flahing fullhaaboot.bin completes successfully. I haven't tried the latest version yet. I was trying v.3.1.0

L4RGER, 2020-09-08:
Have you tried uploading the fullhaaboot from the tasmota front end? What program are you using to flash? Tasmotizer?

wrx5, 2020-09-08:
I flashed tasmota.bin with Tasmotizer and with esptool.py. Both works

I use esptool.py to flash fullhaaboot.bin. I have also used Tasmotier to flash fullhaaboot.bin.

L4RGER, 2020-09-08:
What about uploading fullhaaboot from tasmota front end?

wrx5, 2020-09-08:
what do you mean by tasmota front end? When I flash tasmota I saw upgrade option and used that to flash fullhaaboot.bin. Still doesn't work

L4RGER, 2020-09-08:
Yeh By front end I mean using the browser page to upload the haa bin file. Sounds like you did it. 
Not sure what to tell you, I’ve done about 7 now with no dramas getting the wifi AP

wrx5, 2020-09-08:
let me try the v.3.1.2 and I will let you know

L4RGER, 2020-09-08:
Have you checked a few different devices for the AP? After flashing do you install the switch to mains and try to set up from there or do you do it using the LV from the device you’ve flashed with. 
I’m not 100% sure, but I think the last one I flashed I just reset and plugged the USB back in and used the 3.3v to pair it

L4RGER, 2020-09-08:
Have you joined the discord? A lot of knowledge on there, plus it’s easier to chat

L4RGER, 2020-09-08:
https://discord.gg/v8hyxj2

wrx5, 2020-09-08:
I have only one device. So couldn't try with any other device. 

I use FTDI to try not main. I tried with main but didn't connect the graund. Just live an neutral. I don't know if that would create any issue. because all ground cables are connected together behind the existing switch. and ground is not connected to that switch