RavenSystem/esp-homekit-devices #1121: Power Meter

lfiocco, 2020-09-07:
Hi there, can you please point out some hardware that works with you code? 

Is there some prebuilt we can buy, flash it?

peros550, 2020-09-07:
Blitzwolf SHP6 (the metered version) and those below: 

Example for a Gosund SP111 v1.1 with overvoltage (245VAC) and overcurrent (15A) protection:
{"c":{"l":2,"b":[{"g":13,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":15},{"g":0,"v":1}]},"1":{"r":[{"g":15,"v":1},{"g":0}]},"b":[{"g":13}]},{"t":75,"c0":5,"c1":4,"sl":12,"vf":0.13334,"cf":0.01295,"pf":1.5312,"y0":[{"v":245,"0":{"m":[{"g":1}]}}],"y1":[{"v":15,"0":{"m":[{"g":1}]}}]}]}

Example for a Gosund SP211 with overvoltage (245VAC)and overcurrent (16A) protection:
{"c":{"l":1,"b":[{"g":0,"t":5},{"g":12,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":14},{"g":2,"v":1}]},"1":{"r":[{"g":14,"v":1},{"g":2,"v":0}]},"b":[{"g":12}]}, {"t":2,"0":{"r":[{"g":16},{"g":13,"v":1}]},"1":{"r":[{"g":16,"v":1},{"g":13,"v":0}]},"b":[{"g":0}]},{"t":75,"c0":4,"c1":5,"sl":3,"vf":0.1564,"cf":0.012355,"pf":1.7115,"y0":[{"v":245,"0":{"m":[{"g":1},{"g":2}]}}],"y1":[{"v":16,"0":{"m":[{"g":1},{"g":2}]}}]}]}

WizBangCrash, 2020-09-14:
Hi @lfiocco,
I've been using the new code with these 2nice smart wifi plugs I bought on Amazon
https://smile.amazon.co.uk/2NICE-Socket-Wireless-Control-Required/dp/B07ZQ34X55

Check out the Wiki [here](https://github.com/RavenSystem/esp-homekit-devices/wiki/Power-Measure#measurement-calibration) for how to calibrate them.

closed 2020-10-12 by RavenSystem

wildman199, 2020-10-31:
> Blitzwolf SHP6 (the metered version) and those below:
>
> Example for a Gosund SP111 v1.1 with overvoltage (245VAC) and overcurrent (15A) protection: [...]

Will this work for DC POWER?  I’m looking for a way to monitor my battery.