RavenSystem/esp-homekit-devices #1122: Sonoff POW R2

alifarooq1991, 2020-09-07:
Please anyone share json for sonoff pow with power measurement

d0minikh, 2020-10-07:
+1 would be nice to get it to work :)

RavenSystem, 2020-10-12:
Sorry, it uses UART and it's not supported.

closed 2020-10-12 by RavenSystem

i3laze, 2020-12-12:
Although it might work as a generic HAA-switch, you will loose the metering functionality.
Check Sonoff POW GPIO, if needed:
https://templates.blakadder.com/sonoff_Pow_R2.html

i3laze, 2021-06-14:
Maybe, maybe there could be some inspiration in Tasmota code? 
They support Pow R2 with CSE7766 chip for years:
https://github.com/arendst/Tasmota/blob/5e805e0c4ac1304bfe86a3abd1b6b99b7dda2daa/tasmota/xnrg_02_cse7766.ino

Aaqu, 2021-07-11:
I can become a tester when the release comes out

novellaip, 2021-09-07:
Jumping wire from CSE7759B PIN CF to ESP8285 I05 (I04) can monitor the power only.

i3laze, 2022-01-14:
@RavenSystem still no change of mind on UART? Got two POW R2 units for tests.

tonysprenk, 2022-01-14:
> @RavenSystem still no change of mind on UART? Got two POW R2 units for tests.

Reading serial commands is a confirmed feature coming in HAA 10.0!

rg2k, 2022-03-14:
Any chance this is now working with HAA 10 being released?

MukatP, 2024-01-30:
Sonoff POW 1 v. 2.0 works on HHA 12. in App Home it does not show parameters. Works as a switch. Everything is in the HAA Home Manager app.

i3laze, 2024-09-02:
@RavenSystem , sorry for necroposting.
Amy updates on power measurement via UART since?

vglucas, 2024-09-17:
Script MEPLHAA
Sonoff Pow Power Monitoring Switch Module

https://github.com/RavenSystem/esp-homekit-devices/discussions/2294