RavenSystem/esp-homekit-devices #1123: NodeMCU No Response after updating to 3.1.2

jRaufe, 2020-09-08:
Great job! I have more than 20 devices with HAA, it´s fantastic. Everything updated to 3.1.1 with success except one NodeMCU used to control the watering with 7 valves and two switches, one for OTA update and the other as “enable” button. It worked perfectly until version 2.5.x.

The JSON is
{"c":{"l":2,"b":[{"g":0,"t":5}]},"a":[
{"0":{"r":[{"g":5,"v":1}],"m":[{"g":2,"v":1}]},"1":{"r":[{"g":5}]},"t":20,"w":1,"d":600},
{"0":{"r":[{"g":4,"v":1}],"m":[{"g":3,"v":1}]},"1":{"r":[{"g":4}]},"t":20,"w":1,"d":600},
{"0":{"r":[{"g":3,"v":1}],"m":[{"g":4,"v":1}]},"1":{"r":[{"g":3}]},"t":20,"w":1,"d":600},
{"0":{"r":[{"g":14,"v":1}],"m":[{"g":5,"v":1}]},"1":{"r":[{"g":14}]},"t":20,"w":1,"d":600},
{"0":{"r":[{"g":12,"v":1}],"m":[{"g":6,"v":1}]},"1":{"r":[{"g":12}]},"t":20,"w":1,"d":600},
{"0":{"r":[{"g":13,"v":1}],"m":[{"g":7,"v":1}]},"1":{"r":[{"g":13}]},"t":20,"w":1,"d":600},
{"0":{"r":[{"g":15,"v":1}]},"1":{"r":[{"g":15}]},"t":20,"w":1,"d":1200},
{"t":2},
{"1":{"s":[{"a":1}]},"i":0.5}]}

To get it installed in Homekit I use first this simplier form:
{"a":[{"t":20},{"t":20},{"t":20},{"t":20},{"t":20},{"t":20},{"t":20},{"t":2},{"1":{"s":[{"a":1}]},"i":0.5}]}

With 3.1.1, after installing the reduced form with success, when I use setup mode to change to JSON, I only get NoResponse. I have checked that the device is not in my wifi-net any more.

Any suggestions?
Is there a way to reinstall old versions using only my Mac?

RavenSystem, 2020-09-09:
Try removing status LED `"l":2` declaration. It is a workaround.

jRaufe, 2020-09-09:
Thank you for your answer. I will try it on Saturday when I come back.
But this "l":2 was always in the JSON I’ve used with the older HAA versions and always with success.
I’ll try it anyway, of course.
Is it posible to restore older version using iPad and Mac?

jRaufe, 2020-09-13:
I’ve tried what you said and....it works!
Thank you.

closed 2020-09-13 by jRaufe