RavenSystem/esp-homekit-devices #1124: Support for RGB+CCT (RGBWW, RGB+WW+CW) LED controllers

kevinjohncutler, 2020-09-09:
Not quite a duplicate of #631, #638, #800, as the first one never got resolved (I bet they were successful only for the use of a RGB controller replicating color temperature) and the other two are just for bulbs, which I am guessing are a lot harder to test and debug than [this controller](https://smile.amazon.com/dp/B01DY56N8U?psc=1&ref=ppx_pop_dt_b_product_details). I bought a couple of them and have flashed HAA onto them successfully. I see that in the Wiki there is syntax available for addressing each of the 5 channels, and [this post](https://github.com/RavenSystem/esp-homekit-devices/issues/689#issuecomment-667753190) seems to indicate that it is possible to get it working at present. I'm still learning the json syntax and looking at examples in haa-configurator, but please let me know if I will be toiling in vain haha.

@RavenSystem I saw in #638 that all you need is a better algorithm to mix colors, and this should be easy for me to debug with this controller (I can simply disconnect the leads to isolate whichever LEDs we need to test). Let me know how I can help.

kevinjohncutler, 2020-09-10:
I'm not sure if the json I posted above actually stuck - I thought so because it was giving me different behavior than a different controller, but minor modifications to it no longer would apply (invalid json keeps it in setup mode). However, the following works:

```
{"c":{"n":"LEDENET 5in1 Controller"},"a":[{"t":30,"r":14,"g":12,"v":13,"cw":5,"ww":15}]}
```

This was a modification of the post mentioned in the OP (the json for the "Deta Grid Connect Smart 10W LED Downlight - RGB-WW-CW"). Note that all I had to do was remap the channels to match my device. At first I included the following secondary accessory,

```
,{
		"i":0.5,
		"1":{"s":[{"a":1}]}
	}
```

but I removed it hoping that it was the cause of the latency I am observing. It was not - there is still a lot of lag when selecting a color. Trying to get it to land on a specific color (say, red) causes it to stutter and get stuck on the wrong nearby color (magenta in this case). But the aforementioned "i" and "1" lines above don't affect the behavior. I would very much like some help understanding what this is doing - the documentation doesn't _quite_ get me there.

It would also be informative to try to understand what was wrong with my original json. I used a json verification and formatting tool to put it into compact form and make sure no bad characters made it in etc.

Could this stuttering that I see in the color wheel selection be fixed by changing any of the other parameters (like color transition step), or do you think there is another reason?

jasongoldsmith, 2020-09-11:
@kevinjohncutler Are you in the Discord Server? There's a channel dedicated to this topic there - I can fill you in on my research so far and share some papers on the topic.

The short answer is that the current algorithm is pretty naive with it's assumptions about the LEDs and uses a fairly simplistic mixing algorithm that doesn't take into account perceptive color and a few other things. I'm also looking at the code.

The relevant lightbulb code is in devices/HAA/main.c starting at line 1384.

kevinjohncutler, 2020-09-11:
@jasongoldsmith I'm not, but I'll head over there! Thanks so much for the info.

wayn3, 2020-09-11:
https://discord.gg/v8hyxj2

closed 2020-10-12 by stale[bot]

KoCMoHaBTa, 2024-11-29:
Hi, I'm currently playing with this [MINI-WF-5CH-A _v1.0 MagicHome controller](https://github.com/RavenSystem/esp-homekit-devices/wiki/lightbulb#Automatic-Dimmer-Step). It is RGB-CCT 5 channel controller.

The JSON i'm using is 

```
{"c":{"io":[[[0],6],[[4,5,12,13,14],7]],"b":[[0,5]]},"a":[{"t":30, "s": 1, "g":[5,4,14,13,12],"b":[[0]]}]}
```

So far so good, i have everything I need, but I noticed that 
- when any white color temperature is selected - the RGB channels are not turned off
- when the RGB scale move to the white center - the white channel(s) are turned on

I'm seeing this behavior in led strip controllers for first time - all other controllers i've been using are switching between either RGB channels or CCT channels. 
I would agree that having all 5 channels up is a nice way to achieve more luminance, however this crew up the colors totally.

My question is - is there some sort of configuration i'm missing (cause it's 2:25 AM atm and i need a sleep) or this is intentional or a bug?
If there is no configuration yet:
- is it possible to add some in a future release that would allow us to choose which kind of behavior we want to use
- is it possible to workaround this via some sort of (virtual?) accessory that would turn on/off/set brightness only for specific channel/GPIO?